prophesee-ai/openeb #1: imu support missing

berndpfrommer, 2021-05-21:
I noticed that support for IMU reading has been deprecated. What's the new way then to read the IMU?
Thanks!

laurentbristiel, 2021-05-28:
Hello,

support for IMU reading has indeed been deprecated since Metavision 2.1.0.
So the last version with IMU support is Metavision 2.0.0 and OpenEB 2.2.0 does not support it.

If you used Metavision Software with IMU in the past and want to continue using it, then you should stick with version 2.0.0.
If you willing to use RAW files that contain IMU events produced by Metavision 2.0.0 (or earlier) with another software/script, we can send you the format of the IMU events so that you can decode this file and use it as you wish.

Hope this helps!

closed 2021-05-28 by laurentbristiel

berndpfrommer, 2021-05-28:
Is there any way to access the IMU in real-time? Or are you stopping support for real-time IMU completely going forward?
Thanks!

laurentbristiel, 2021-05-28:
We stopped real-time IMU support since 2.1.0 and, fow now, don't plan to support it again in the future.