prophesee-ai/openeb #2: is the nvidia xavier supported?

berndpfrommer, 2021-05-31:
Hi,
I was looking into buying a SilkyEVCam to be used on a robot that runs an Nvidia Xavier AGX (ARM64 based system). Is that supported by OpenEB? Will it be possible to get a HAL plugin for ARM64/Ubuntu 18.04 systems?
Thanks!

berndpfrommer, 2021-05-31:
Alright, compilation of OpenEB on NVidia Xavier AGX under Ubuntu 18.04 worked first shot without warnings or error!
This still leaves the issue of requiring a HAL plugin. I'm planning on buying a SilkyEVCam, but currently have the following camera on hand:
`Serial number: Prophesee:hal_plugin_gen31_fx3:00001753`
Is there a way to get source code or an ARM64 plugin for that? I will also contact Prophesee directly about this.

laurentbristiel, 2021-05-31:
Hi Bernd,
Great to read that you were able to compile OpenEB on NVidia Xavier AGX under Ubuntu 18.04! 
So for the camera plugins:
- for your Prophesee EVK Gen31, you can contact directly (https://www.prophesee.ai/contact-us/) and we will provide you with the binaries compiled for NVidia Xavier AGX under Ubuntu 18.04
- for your future SilkyEVCam, the plugin is managed by SilkyEVCam vendor, so please contact them to get the binaries for Xavier. If it does not succeed, come back to us on this and we will help you in this process.

closed 2021-05-31 by laurentbristiel

berndpfrommer, 2021-05-31:
Laurent,
Mercy beaucoup!

mandulaj, 2022-11-01:
Hi all,
I am also trying to use the OpenEB v3.1.0 on the Nvidia Xavier NX. Like @berndpfrommer, I have managed to compile everything first shot without errors and can play prerecorded files without issue using the `metavision_viewer`

However, I am having issues reading the live image from the Prophesee EVK4 camera. I wanted to ask @berndpfrommer, if you managed to connect your camera successfully.

------

The board I am using is the [ AAEON BOXER-8251AI](https://www.aaeon.com/en/p/nvidia-xavier-nx-embedded-box-pc-boxer-8251ai). I have attached the camera to a USB3.2 Gen 1 port and it does show up in the Kernel Log correctly:

```
Nov 01 15:12:10 aaeon-desktop kernel: usb 2-3.1: new SuperSpeed USB device number 10 using tegra-xusb
Nov 01 15:12:10 aaeon-desktop kernel: usb 2-3.1: LPM exit latency is zeroed, disabling LPM.
Nov 01 15:12:10 aaeon-desktop kernel: usb 2-3.1: New USB device found, idVendor=04b4, idProduct=00f5
Nov 01 15:12:10 aaeon-desktop kernel: usb 2-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Nov 01 15:12:10 aaeon-desktop kernel: usb 2-3.1: Product: EVK4
Nov 01 15:12:10 aaeon-desktop kernel: usb 2-3.1: Manufacturer: Prophesee
Nov 01 15:12:10 aaeon-desktop mtp-probe[20684]: checking bus 2, device 10: "/sys/devices/3610000.xhci/usb2/2-3/2-3.1"
Nov 01 15:12:10 aaeon-desktop mtp-probe[20684]: bus: 2, device: 10 was not an MTP device
```

Under `lsusb` it shows up as:
```bash
lsusb # On Xavier
Bus 002 Device 011: ID 04b4:00f5 Cypress Semiconductor Corp. 
```
which is slightly different from the way it shows up on my Dell laptop x86 machine (where the camera works without issues)
```bash
lsusb # On Dell laptop
Bus 002 Device 007: ID 04b4:00f5 Cypress Semiconductor Corp. EVK4
```

But running `metavision_viewer` gives me this error:

```
Simple viewer to stream events from a RAW file or device, using the SDK driver API.
Press SPACE key while running to record or stop recording raw data
Press 'q' or Escape key to leave the program.
Press 'r' to toggle the hardware ROI given as input.
Press 'e' to toggle the ERC module (if available).
Press '+' to increase the ERC threshold (if available).
Press '-' to decrease the ERC threshold (if available).
Press 'h' to print this help.
 

------------------------------------------------
Metavision SDK Driver exception

Error 101001: Camera not found. Check that a camera is plugged on your system and retry.
```

I don't think it was required, but I sourced the `utils/scripts/setup_env.sh` anyways without any success. If I understand correctly, since OpenEB v3.0.0, there is no need to include special plugins for a specific camera.

My questions thus are:
* Is there any other method of reading the Prophesee camera besides the Closed source drivers (which are x86 only). 
* Can I get some more detailed error messages in order to help debug the issue?
* 
I would appreciate any suggestions and help on how to proceed.

mandulaj, 2022-11-01:
Also @laurentbristiel, perhaps I can open this as a new issue?

berndpfrommer, 2022-11-01:
@mandulaj alas I do not have the EVK4 hardware.
There are two reasons I can think of why it wouldn't open the camera:
1) Device permissions set wrong. No idea how to fix that on a USB device, but you could try running as root first to see if that is possibly the issue.
2) hardware-dependent plugin not found. I have had good luck with the linux command "strace -f  metavision_viewer >& /tmp/foo.txt". It will show you all system calls the program makes, including what files are opened, so you can check if it finds the HAL plugin, and where it's looking for it.

mandulaj, 2022-11-01:
Hey @berndpfrommer , thanks a lot for getting back to me. Those are some good suggestions. I will try them out first thing tomorrow.

mandulaj, 2022-11-02:
Yes @berndpfrommer, that was indeed an issue. I added a udev rule with read permissions to all users on the USB and now it works perfectly! Thanks a lot for the suggestion. 

I will add the solution to the issue I opened in #65.

berndpfrommer, 2022-11-02:
As an action item for the OpenEB maintainers: a proper error message could have prevented this issue from being filed. Usually system calls to open the device returned EPERM, so that could be translated into a more indicative error message than "Camera not found".

mandulaj, 2022-11-02:
Agree,  the only way I found out was by using `strace` and seeing a `Permission Denied` when the USB device was accessed.

nbourdis-psee, 2022-11-07:
Thank you for your feedback @berndpfrommer @mandulaj, we will try to improve this in coming releases.