prophesee-ai/openeb #3: Errors when performing compilation tests

juanmed, 2021-08-11:
Hi,

Thank you for releasing this framework. I tried to compile and install openeb in my Ubuntu18.04 pc.I followed the instructions in this repository. Compilation goes fine but when I do ```ctest --verbose``` I get many errors. The summary is:

```
38% tests passed, 13 tests failed out of 21

Total Test time (real) =   9.39 sec

The following tests FAILED:
	  1 - metavision-raw-cutter-test (Failed)
	  4 - metavision-raw-info-with-sample-plugin-test (Failed)
	  5 - metavision-raw-cutter-with-sample-plugin-test (Failed)
	  6 - metavision-raw-to-csv-with-sample-plugin-test (Failed)
	  7 - metavision-raw-to-dat-with-sample-plugin-test (Failed)
	 11 - pytests_base (Failed)
	 13 - metavision-raw-to-video-test (Failed)
	 15 - pytests_core (Failed)
	 16 - metavision-raw-info-test (Failed)
	 17 - metavision-raw-to-dat-test (Failed)
	 18 - metavision-raw-to-csv-test (Failed)
	 20 - metavision-evt2-raw-file-decoder-test (Failed)
	 21 - metavision-evt3-raw-file-decoder-test (Failed)

```
Here is the full output of the ctest command: 
[test.log](https://github.com/prophesee-ai/openeb/files/6967424/test.log)

I will appreciate if you could guide me on how to fix this errors.

KingArrkinian, 2021-08-11:
Hello Juan,

Did you install the HAL Plugins and the datasets? It's explained in the lower part of the README that you need those for the tests to run.

https://github.com/prophesee-ai/openeb#optional-running-the-test-suite

You might also want to install opencv-python.

juanmed, 2021-08-12:
Hi. Yes, I installed the HAL plugins and downloaded/unzip the dataset as
explained in the README.

nbourdis-psee, 2021-08-12:
Hi Juan,

The README seem to be missing one step to be done after the build has completed and before running tests or samples: setup the execution environment, i.e. add specific include and library directories to the PATH. To make this easy, the build tree generates two `setup_env` scripts, one for Linux and one for Windows, in the `utils/scripts` folder of the build directory.

So in your case on Ubuntu, once the build completes successfully, simply source this script in your terminal session before running the tests: `source ./utils/scripts/setup_env.sh`. Note that this will need to be done for each new terminal session.

Hope this helps,
Nicolas

KingArrkinian, 2021-08-12:
Hello,

I believe the datasets were not placed correctly, since you have these assert os.path.exists(file) errors:

1:         # To be sure the error isn't thrown because the input file doens't exist, use one of the datasets
1:         input_rawfile = os.path.join(dataset_dir, "gen31_timer.raw")
1: >       assert os.path.exists(input_rawfile)

Just in case, are your datasets in <OPENEB_SRC_DIR>/datasets/ or <OPENEB_SRC_DIR>/datasets/datasets/?

Unzipping creates a datasets directory already, so if you unzipped them at <OPENEB_SRC_DIR>/datasets/, it might have created a subdirectory, which will not work.

laurentbristiel, 2021-08-24:
I am assuming @KingArrkinian comment helped @juanmed => closing this issue
If this is not the case, don't hesitate to ask for more help.

closed 2021-08-24 by laurentbristiel