prophesee-ai/openeb #4: Errors during compiling OpenEB on Ubuntu 18.04

kamzero, 2021-08-19:
Hello,

I followed the instructions in this repository to install OpenEB on my Ubuntu 18.04. But I got errors after I did `cmake --build . --config Release -- -j 4`. Here is a part of the summary:

`/home/kamzero/Downloads/openeb/sdk/modules/core/cpp/src/video_writer.cpp: In member function ‘virtual void Metavision::VideoWriter::write(cv::InputArray)’:
/home/kamzero/Downloads/openeb/sdk/modules/core/cpp/src/video_writer.cpp:343:33: error: no matching function for call to ‘Metavision::VideoWriter::write(const cv::_InputArray&)’
     cv::VideoWriter::write(image);
                                 ^
In file included from /home/kamzero/Downloads/openeb/sdk/modules/core/cpp/src/video_writer.cpp:51:0:
/usr/local/include/opencv2/videoio.hpp:940:26: note: candidate: virtual void cv::VideoWriter::write(const cv::Mat&)
     CV_WRAP virtual void write(const Mat& image);
                          ^~~~~
/usr/local/include/opencv2/videoio.hpp:940:26: note:   no known conversion for argument 1 from ‘const cv::_InputArray’ to ‘const cv::Mat&’
/home/kamzero/Downloads/openeb/sdk/modules/core/cpp/src/video_writer.cpp: In member function ‘virtual Metavision::VideoWriter& Metavision::VideoWriter::operator<<(const cv::UMat&)’:
/home/kamzero/Downloads/openeb/sdk/modules/core/cpp/src/video_writer.cpp:365:72: error: no matching function for call to ‘Metavision::VideoWriter::operator<<(const cv::UMat&)’
     return static_cast<VideoWriter &>(cv::VideoWriter::operator<<(image));
                                                                        ^
In file included from /home/kamzero/Downloads/openeb/sdk/modules/core/cpp/src/video_writer.cpp:51:0:
/usr/local/include/opencv2/videoio.hpp:931:26: note: candidate: virtual cv::VideoWriter& cv::VideoWriter::operator<<(const cv::Mat&)
     virtual VideoWriter& operator << (const Mat& image);
                          ^~~~~~~~
/usr/local/include/opencv2/videoio.hpp:931:26: note:   no known conversion for argument 1 from ‘const cv::UMat’ to ‘const cv::Mat&’
sdk/modules/core/cpp/lib/CMakeFiles/metavision_sdk_core.dir/build.make:185: recipe for target 'sdk/modules/core/cpp/lib/CMakeFiles/metavision_sdk_core.dir/__/src/video_writer.cpp.o' failed
make[2]: *** [sdk/modules/core/cpp/lib/CMakeFiles/metavision_sdk_core.dir/__/src/video_writer.cpp.o] Error 1
CMakeFiles/Makefile2:2203: recipe for target 'sdk/modules/core/cpp/lib/CMakeFiles/metavision_sdk_core.dir/all' failed
make[1]: *** [sdk/modules/core/cpp/lib/CMakeFiles/metavision_sdk_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 35%] Built target metavision_sdk_base_python3
[ 36%] Linking CXX executable ../../../../bin/metavision_raw_cutter
[ 36%] Built target metavision_raw_cutter
[ 37%] Linking CXX executable ../../../../bin/metavision_platform_info
[ 37%] Built target metavision_platform_info
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2
`

It would be greatly appreciated if you can help.

ValeHaas, 2021-08-19:
Currently seeing the same behaviour during compilation, the relevant part seems to be:

```
[ 23%] Building CXX object hal/cpp/lib/CMakeFiles/metavision_hal.dir/__/src/utils/demangle.cpp.o
/build/openeb/sdk/modules/core/cpp/src/video_writer.cpp: In member function ‘virtual void Metavision::VideoWriter::write(cv::InputArray)’:
/build/openeb/sdk/modules/core/cpp/src/video_writer.cpp:343:28: error: cannot convert ‘const cv::_InputArray’ to ‘const cv::Mat&’
  343 |     cv::VideoWriter::write(image);
      |                            ^~~~~
      |                            |
      |                            const cv::_InputArray
In file included from /build/openeb/sdk/modules/core/cpp/src/video_writer.cpp:51:
/usr/local/include/opencv2/videoio.hpp:950:43: note:   initializing argument 1 of ‘virtual void cv::VideoWriter::write(const cv::Mat&)’
  950 |     CV_WRAP virtual void write(const Mat& image);
      |                                ~~~~~~~~~~~^~~~~
/build/openeb/sdk/modules/core/cpp/src/video_writer.cpp: In member function ‘virtual Metavision::VideoWriter& Metavision::VideoWriter::operator<<(const cv::UMat&)’:
/build/openeb/sdk/modules/core/cpp/src/video_writer.cpp:365:67: error: cannot convert ‘const cv::UMat’ to ‘const cv::Mat&’
  365 |     return static_cast<VideoWriter &>(cv::VideoWriter::operator<<(image));
      |                                                                   ^~~~~
      |                                                                   |
      |                                                                   const cv::UMat
In file included from /build/openeb/sdk/modules/core/cpp/src/video_writer.cpp:51:
/usr/local/include/opencv2/videoio.hpp:941:50: note:   initializing argument 1 of ‘virtual cv::VideoWriter& cv::VideoWriter::operator<<(const cv::Mat&)’
  941 |     virtual VideoWriter& operator << (const Mat& image);
      |                                       ~~~~~~~~~~~^~~~~
```
[image]

laurentbristiel, 2021-08-24:
Hello @kamzero @ValeHaas 
You both seem to have issues with the OpenCV version available on your Ubuntu 18.
When you follow the instruction of the README, OpenCV should be installed with this statement:
```
sudo apt -y install cmake libopencv-dev git
```

You should end-up with version 3.2 of OpenCV. You can check it this way:

```
dpkg -l  | grep libopencv-dev
libopencv-dev    3.2.0+dfsg-4ubuntu0.1    amd64   development files for opencv
```

If this is not the case, you could try to remove/re-install OpenCV:
```
sudo apt remove "libopencv*" "opencv-*" --purge
sudo apt -y install libopencv-dev
```

And try again the compilation step.

If this still fails, please share:
- the output of the step `cmake .. -DBUILD_TESTING=OFF` (or your equivalent "cmake .." call)
- the output of `dpkg -l  | grep libopencv-dev`

ValeHaas, 2021-08-24:
Hello @laurentbristiel , thank you for your reply.

My issue came from a conflicting version of open cv. I had opencv v3.4.15 installed by compiling it from source. 
Removing it and re-installing as described above solved the issue.

For completeness, my outputs of the commands above still looked as described.

laurentbristiel, 2021-08-24:
Thanks for the update @ValeHaas 
I will assume that @kamzero problem had a similar root cause and close this issue.

closed 2021-08-24 by laurentbristiel