prophesee-ai/openeb #7: what's the meanning of the function "make_moving_edgelet_in_time_surface"

zebrajiahao, 2021-08-28:
I learned the metavision_edgelet_2d_tracking, and want to know how this function work? 
Such as the formulation is to do in the below pic:
[image]

Some papers could be suggested to read.
Looking forward to your reply,Thx.

laurentbristiel, 2021-08-30:
Hello,
the function you are refering to is not part of OpenEB but part of [Metavision Professional](https://www.prophesee.ai/metavision-intelligence-professional-request/) so I assume you are a Prophesee Customer using this Pro code. If so, you are granted some support in your package and can contact support@prophesee.ai for your question.
If your context is different, please explain it.

closed 2021-08-30 by laurentbristiel

laurentbristiel, 2021-09-15:
We had no feedback from you and the code you shared is private, so I am deleting this issue to not expose copyrighted code.
Please contact us again if you need support but don't shre publicly copyrighted code.