prophesee-ai/openeb #16: Protecting the main and develop branches

KingArrkinian, 2021-09-14:
Hello,

I earlier committed erroneously on the main branch here instead of on my fork.
While the mistake was corrected with:
`git reset --hard $former_commit_sha
git push --force`
it is a common mistake that is bound to happen when the folders on both the base and the fork have the same name, and other contributors with writing rights will probably commit this error again.

I'd like to suggest that we block both main and develop from any direct commit, using Github's protected branch workflow.
https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/defining-the-mergeability-of-pull-requests/about-protected-branches

This would not apply to admins of the repository, but only to users with **maintainer** or below access level.

Blocking develop is not as important as main, but it will enforce merge reviewability.

laurentbristiel, 2021-09-14:
Looks good to me. As far as I know, this is the way our git repo (like metavision) are configured internally currently.

cbernard-psee, 2021-09-14:
Thanks @KingArrkinian  for point out this issue. Settings updated as follow for main and develop branches
[image]

KingArrkinian, 2021-09-14:
[image]
Direct input is now prevented, and merges will be secured by the CI.
It looks great, I'm closing this issue.