prophesee-ai/openeb #32: Failure to build install on Ubuntu

Jegp, 2021-10-16:
I'm trying to build and install openeb for use as a third-party library. Running `sudo cmake --build . --target install` generates the following error:

```
CMake Error at sdk/modules/core/python/pypkg/cmake_install.cmake:50 (file):
  file INSTALL given no DESTINATION
Call Stack (most recent call first):
  sdk/modules/core/python/cmake_install.cmake:49 (include)
  sdk/modules/core/cmake_install.cmake:48 (include)
  sdk/modules/cmake_install.cmake:48 (include)
  sdk/cmake_install.cmake:59 (include)
  cmake_install.cmake:78 (include)
```

I'm on a standard Ubuntu 21.04 machine. I followed the prerequisite installations, so all dependencies should be up to date. I can reproduce the same behaviour with cmake 3.18.4 and 3.22.0-rc1.

Thanks for your time!

Jegp, 2021-10-18:
Thanks for your quick reply!

I changed `cmake/custom_functions/python3.cmake` as you suggested (see diff below), but I arrived at the same error. Even after a fresh clone. However, I think the default version of Python in Ubuntu 21 is `3.9`, and fixing that in the cmake file fixed it!

Here's my diff if that's helpful. Happy to wrap it up in a PR, but I imagine it's helpful to add a check like you did for focal instead of just hardcoding it in.

```diff --git a/cmake/custom_functions/python3.cmake b/cmake/custom_functions/python3.cmake
index 4a89256..5f709b1 100644
--- a/cmake/custom_functions/python3.cmake
+++ b/cmake/custom_functions/python3.cmake
@@ -116,12 +116,12 @@ if (COMPILE_PYTHON3_BINDINGS)

     # this variable is used to create all python versions packages variables for cpack
     # but not all of them will be generated, only the one indicated by PYBIND11_PYTHON_VERSIONS
-    set (PYTHON3_ALL_VERSIONS "3.6;3.7;3.8")
+    set (PYTHON3_ALL_VERSIONS "3.6;3.7;3.8;3.9")

     # this variable is used to set the default version for package dependency, i.e this version
     # is always available for the current installation
     if (UNIX AND NOT APPLE)
-        set (PYTHON3_DEFAULT_VERSION "3.6")
+        set (PYTHON3_DEFAULT_VERSION "3.9")
         find_program(_lsb_release_exec lsb_release)
         if (_lsb_release_exec)
             execute_process(COMMAND ${_lsb_release_exec} -cs
```

laurentbristiel, 2021-10-19:
Good to read that you were to compile successfully, thanks for the feedback.
I did not realize that Ubuntu 21 was shipping with Python 3.9.
We currently don't support Python 3.9, so it's good that it compiles but I don't guarantee that the whole Python API works OK. We will see internally how/when we want to work on the support of Python 3.9 and in the meantime we prefer to not change the code we both discussed as only 3.6, 3.7 and 3.8 are supported. 

Note that the diff you shared could help future users working on Ubuntu 21 anyway. Thanks for that!

closed 2021-10-19 by laurentbristiel