prophesee-ai/openeb #36: Ubuntu installation failed: python not found

shiba24, 2021-12-10:
Hi, thank you for your support. I have two questions about python in compilation.

# Environment

```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
```
## python

I have python3.8 and 3.9:

```
➜  build git:(main) ✗ which python
python not found
➜  build git:(main) ✗ which python3
/usr/bin/python3
➜  build git:(main) ✗ which python3.8
/usr/bin/python3.8
➜  build git:(main) ✗ which python3.9
/usr/bin/python3.9
➜  build git:(main) ✗ python3 -V
Python 3.8.10
```

# Q1. Compilation fails because it does not find python3.8

I think cmake tries to find python3.9 first although I have 3.8. How can I tell the pythonpath and/or python version?

Here is the error:

```
build git:(main) ✗ cmake .. -DBUILD_TESTING=OFF
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building METAVISION_SDK_MODULES_OPEN modules : base;core;driver;ui
-- Building METAVISION_SDK_MODULES_PROFESSIONAL modules : analytics;calibration;cv;ml;3dview;cv3d
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Found LibUSB: /usr/include/libusb-1.0  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: program_options filesystem timer chrono thread 
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Python3_LIBRARIES
  Development Development.Module Development.Embed) (found version "3.9.7")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindPython/Support.cmake:2999 (find_package_handle_standard_args)
  cmake/Modules/FindPython3.cmake:391 (include)
  cmake/custom_functions/python3.cmake:63 (find_package)
  CMakeLists.txt:211 (include)

-- Configuring incomplete, errors occurred!
See also "/home/shintaro/prophesee_ws/openeb/build/CMakeFiles/CMakeOutput.log".
See also "/home/shintaro/prophesee_ws/openeb/build/CMakeFiles/CMakeError.log".

```

I tried this issue but didn't work either. https://github.com/prophesee-ai/openeb/issues/32#issuecomment-945825601

# Q2. virtual environment

My existing python environment is already coupled with ROS and some DAVIS packages. Since your library requirement is somehow strict (`"numpy==1.19.5" "opencv-python>=4.2.0.34"`), I'd like to create virtual environment to encapsulate openeb environment. 
I'm thinking something like `python3.8 -m venv ~/local/venv` but poetry is also fine.

Have you every tried it? How can I make it work with venv?

Sincerely,

Shintaro

laurentbristiel, 2021-12-15:
Hello Shintaro,

sorry for the delay to answer your message.
So it looks indeed that CMake is selecting Python 3.9. Even if we don't officially support this version of Python, I am surprised that the compilation fails. Are you sure that Python 3.9 is properly installed, and specifically, are the headers available? (you should not have the error `missing: Python3_INCLUDE_DIRS Python3_LIBRARIES`). To fix this issue and force Ptyhon 3.8 to be used, using Virtual Environment looks like a right choice. There is nothing special to do besided the `Python3.8 -m venv ~/local/venv` you mention. Doesn't the compilation work if you follow the prerequisites and compile in this env?

shiba24, 2022-01-07:
Hey @laurentbristiel 
thank you for the reply. But the problem is not resolved by your reply.

> Are you sure that Python 3.9 is properly installed

Yes, because I can use it without any problems.

> and specifically, are the headers available?

Could you tell me how I can check it?

> To fix this issue and force Ptyhon 3.8 to be used, using Virtual Environment looks like a right choice. There is nothing special to do besided the Python3.8 -m venv ~/local/venv you mention. Doesn't the compilation work if you follow the prerequisites and compile in this env?

Even I use venv python3.8, the compilation does NOT work. My problem here is even I am in the 3.8 venv, the cmake tries to find 3.9 and fails with the above error that I attached.

```
(venv-prophesee) ➜  build git:(main) ✗ python -V
Python 3.8.10
(venv-prophesee) ➜  build git:(main) ✗ python3 -V
Python 3.8.10
(venv-prophesee) ➜  build git:(main) ✗ cmake .. -DBUILD_TESTING=OFF
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building METAVISION_SDK_MODULES_OPEN modules : base;core;driver;ui
-- Building METAVISION_SDK_MODULES_PROFESSIONAL modules : analytics;calibration;cv;ml;3dview;cv3d
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Found LibUSB: /usr/include/libusb-1.0  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: program_options filesystem timer chrono thread 
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Python3_LIBRARIES
  Development Development.Module Development.Embed) (found version "3.9.7")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindPython/Support.cmake:2999 (find_package_handle_standard_args)
  cmake/Modules/FindPython3.cmake:391 (include)
  cmake/custom_functions/python3.cmake:63 (find_package)
  CMakeLists.txt:211 (include)

-- Configuring incomplete, errors occurred!
See also "/home/shintaro/prophesee_ws/openeb/build/CMakeFiles/CMakeOutput.log".
See also "/home/shintaro/prophesee_ws/openeb/build/CMakeFiles/CMakeError.log".
```

Could you please help me?

laurentbristiel, 2022-01-17:
>> and specifically, are the headers available?
> Could you tell me how I can check it?

On my Ubuntu 18, I checked it this way
1) in Python shell I type
```
from sysconfig import get_paths
get_paths()
```
2) I get this:
```
{'stdlib': '/usr/lib/python3.6', 'platstdlib': '/usr/lib/python3.6', 'purelib': '/usr/lib/python3.6/site-packages', 'platlib': '/usr/lib/python3.6/site-packages', 'include': '/usr/include/python3.6m', 'platinclude': '/usr/include/python3.6m', 'scripts': '/usr/bin', 'data': '/usr'}
```
3) and indeed, in `/usr/include/python3.6m` I do have the include files (.h files)
=> could you check that you get those includes files this way?

shiba24, 2022-01-17:
I think I get those correctly.

```
➜  openeb git:(main) ✗ python3
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from sysconfig import get_paths
>>> get_paths()
{'stdlib': '/usr/lib/python3.8', 'platstdlib': '/usr/lib/python3.8', 'purelib': '/usr/lib/python3.8/site-packages', 'platlib': '/usr/lib/python3.8/site-packages', 'include': '/usr/include/python3.8', 'platinclude': '/usr/include/python3.8', 'scripts': '/usr/bin', 'data': '/usr'}
>>> exit()
➜  openeb git:(main) ✗ ls /usr/include/python3.8/
abstract.h         dynamic_annotations.h  longintrepr.h      pyconfig.h     pythread.h
asdl.h             enumobject.h           longobject.h       pyctype.h      pytime.h
ast.h              errcode.h              marshal.h          py_curses.h    rangeobject.h
bitset.h           eval.h                 memoryobject.h     pydebug.h      setobject.h
bltinmodule.h      fileobject.h           methodobject.h     pydtrace.h     sip.h
boolobject.h       fileutils.h            modsupport.h       pyerrors.h     sliceobject.h
bytearrayobject.h  floatobject.h          moduleobject.h     pyexpat.h      structmember.h
bytes_methods.h    frameobject.h          mpi4py             pyfpe.h        structseq.h
bytesobject.h      funcobject.h           namespaceobject.h  pyhash.h       symtable.h
cellobject.h       genobject.h            node.h             pylifecycle.h  sysmodule.h
ceval.h            graminit.h             numpy              pymacconfig.h  token.h
classobject.h      grammar.h              object.h           pymacro.h      traceback.h
codecs.h           Imaging.h              objimpl.h          pymath.h       tracemalloc.h
code.h             ImagingUtils.h         odictobject.h      pymem.h        tupleobject.h
compile.h          ImDib.h                opcode.h           pyport.h       typeslots.h
complexobject.h    ImPlatform.h           osdefs.h           pystate.h      ucnhash.h
context.h          import.h               osmodule.h         pystrcmp.h     unicodeobject.h
cpython            internal               parsetok.h         pystrhex.h     warnings.h
datetime.h         interpreteridobject.h  patchlevel.h       pystrtod.h     weakrefobject.h
descrobject.h      intrcheck.h            picklebufobject.h  Python-ast.h
dictobject.h       iterobject.h           pyarena.h          Python.h
dtoa.h             listobject.h           pycapsule.h        pythonrun.h

```

shiba24, 2022-01-18:
@laurentbristiel  could you try creating virtual env python, and use it for the compilation? If you can make it with virtual env, it means you get how to specify the venv python environment in cmake.

shiba24, 2022-01-28:
Hey any update on this?

laurentbristiel, 2022-01-30:
Sorry, did not have time to test it myself yet. I discussed it around and people were able to successfuly do what you are trying to do. I need to gather the details and try it myself to understand why you are unable to do it.

laurentbristiel, 2022-02-10:
Hello @shiba24 
Sorry for taking so long to come back to you.
Indeed, I tried on Ubuntu20 with both Python 3.8 and Python3.9 installed and cmake chooses Python3.9 by default.
To force cmake to use another Python version, we can use `-DPython3_EXECUTABLE=<path_to_python38>`

So in our case, this is what we get:
```
$ which python3
/usr/bin/python3
$ which python3.8
/usr/bin/python3.8
$ which python3.9
/usr/bin/python3.9
$ cmake .. -DBUILD_TESTING=OFF -DPython3_EXECUTABLE=/usr/bin/python3.8
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building METAVISION_SDK_MODULES_OPEN modules : base;core;driver;ui
-- Building METAVISION_SDK_MODULES_PROFESSIONAL modules : analytics;calibration;cv;ml;3dview;cv3d
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Found LibUSB: /usr/include/libusb-1.0  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: program_options filesystem timer chrono thread 
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter Development Development.Module Development.Embed 
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/local/include (found version "2.6.0" )
-- Found OpenCV: /usr (found version "4.2.0") found components: core highgui imgproc videoio imgcodecs calib3d objdetect 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem system 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
Git LFS is not available, skipping download of datasets/metavision_core/csv_zip_eth
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.1.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lbristiel/Development/openeb/build
```

Hope this helps!

shiba24, 2022-02-11:
That made my day. It worked for me too! Thank you so much!

closed 2022-02-11 by shiba24