prophesee-ai/openeb #38: Support for OpenEmbedded/Yocto or Buildroot

voinul4, 2021-12-15:
Feature Request!

My team has been successful working with the EVK 2 HD on an Ubuntu desktop. However, our end goal requires us to use a low-power, embedded platform that cannot run Ubuntu.

Are there any plans to support embedded flows like Yocto or Buildroot for OpenEB?

laurentbristiel, 2021-12-17:
Hello,
Thanks for your interest in OpenEB.
We are indeed studying this embedded platform topic. We already did some conclusive tests with Yocto but we don't have a release date yet for official support.
We will keep the community informed when we make progress on this suject.
Regards,
Laurent

hlarouss, 2024-02-22:
Hello,

I'm coming back on this matter to see if any progress was performed. We also plan to interface the Prophesee sensor with an embedded platform that runs on a Yocto-generated Linux OS.

Can we easily compile the latest OpenEB version for an ARM-architecture target using Yocto?

Thanks in advance, for your help,

Kind Regards,
Houssem

leonel-psee, 2024-02-22:
Hello Houssem, 

Actually, since the release 3.0.0 of [OpenEB](https://github.com/prophesee-ai/openeb/releases) (31/05/2022), the source code of Prophesee plugins is distributed.

Therefore, you should be able to compile and use your EVK2 HD (embedding sensor Gen4.1 or IMX636) with OpenEB (including the latest release) on any platform.

PS: If you have an EVK2 with Gen4.0, you could need additional support. Please [contact Prophesee support](https://support.prophesee.ai/portal/en/newticket) in that case.

Best regards,

Leonel

nbourdis-psee, 2024-02-23:
Hi Houssem, just to complement on Leonel's reply, the source code of the plugins is available since version 3.0.0, and we validate cross-compilation to ARM platforms since version 4.5.0. If you plan to connect your camera through USB, then you should be able to take it from there without problems.

Also related to your question about Yocto, we do not yet have a Yocto project bundle with OpenEB ready to be shared, but we are actively working on this and should be able to release a sample one within a couple of months.