prophesee-ai/openeb #39: where can I find the BaseFrameGenerationAlgorithm

sirius112358, 2022-01-22:
I want to run **Blinking Lights detector** samples in local PyCharm IDE.
The OpenEB library has been installed and compiled.
However, I can not find **BaseFrameGenerationAlgorithm** in the **metavision_sdk_core** module. 

Could you help me to solve this problem?
[image]

laurentbristiel, 2022-01-23:
Hi! To get the Python API, you need to compile the Python Binding. For this you need to install pybind (see [README](https://github.com/prophesee-ai/openeb/blob/main/README.md)). Did you install pybind? Note also that you should notinclude DCOMPILE_PYTHON3_BINDINGS=OFF during the "cmake .." otherwise the Python binding are not built.

sirius112358, 2022-01-23:
Hi! Thanks for your reply.
I think I have installed pybind.
Here is the result when I run the command in cmd
[image]

and this is the installed file of vcpkg
[image]

Then I tried to compile openEB using CMake. The command is 
_cmake -A x64 -DCMAKE_TOOLCHAIN_FILE=D:\Prophesee\openeb-2.3.0\cmake\toolchains\vcpkg.cmake -DVCPKG_DIRECTORY=D:\vcpkg-master .._

and here is what I got

> D:\Prophesee\openeb-2.3.0\build>cmake -A x64 -DCMAKE_TOOLCHAIN_FILE=D:\Prophesee\openeb-2.3.0\cmake\toolchains\vcpkg.cmake -DVCPKG_DIRECTORY=D:\vcpkg-master ..
> -- Building for: Visual Studio 17 2022
> -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19043.
> -- The C compiler identification is MSVC 19.30.30528.0
> -- The CXX compiler identification is MSVC 19.30.30528.0
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Check for working C compiler: D:/Microsoft Visual Studio/2022/Preview/VC/Tools/MSVC/14.30.30528/bin/Hostx64/x64/cl.exe - skipped
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: D:/Microsoft Visual Studio/2022/Preview/VC/Tools/MSVC/14.30.30528/bin/Hostx64/x64/cl.exe - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Building METAVISION_SDK_MODULES_OPEN modules : base;core;driver;ui
> -- Building METAVISION_SDK_MODULES_PROFESSIONAL modules : analytics;calibration;cv;ml;3dview;cv3d
> -- Looking for pthread.h
> -- Looking for pthread.h - not found
> -- Found Threads: TRUE
> -- Found LibUSB: D:/vcpkg-master/installed/x64-windows/include/libusb-1.0
> CMake Warning at D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1369 (message):
>   New Boost version may have incorrect or missing dependencies and imported
>   targets
> Call Stack (most recent call first):
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1492 (_Boost_COMPONENT_DEPENDENCIES)
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:2102 (_Boost_MISSING_DEPENDENCIES)
>   CMakeLists.txt:207 (find_package)
> 
> 
> CMake Warning at D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1369 (message):
>   New Boost version may have incorrect or missing dependencies and imported
>   targets
> Call Stack (most recent call first):
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1492 (_Boost_COMPONENT_DEPENDENCIES)
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:2102 (_Boost_MISSING_DEPENDENCIES)
>   CMakeLists.txt:207 (find_package)
> 
> 
> CMake Warning at D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1369 (message):
>   New Boost version may have incorrect or missing dependencies and imported
>   targets
> Call Stack (most recent call first):
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1492 (_Boost_COMPONENT_DEPENDENCIES)
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:2102 (_Boost_MISSING_DEPENDENCIES)
>   CMakeLists.txt:207 (find_package)
> 
> 
> CMake Warning at D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1369 (message):
>   New Boost version may have incorrect or missing dependencies and imported
>   targets
> Call Stack (most recent call first):
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1492 (_Boost_COMPONENT_DEPENDENCIES)
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:2102 (_Boost_MISSING_DEPENDENCIES)
>   CMakeLists.txt:207 (find_package)
> 
> 
[... 44 lines cut ...]
> 
> 
> CMake Warning at D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1369 (message):
>   New Boost version may have incorrect or missing dependencies and imported
>   targets
> Call Stack (most recent call first):
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:1492 (_Boost_COMPONENT_DEPENDENCIES)
>   D:/CMake/share/cmake-3.22/Modules/FindBoost.cmake:2102 (_Boost_MISSING_DEPENDENCIES)
>   utils/cpp/profiling/lib/CMakeLists.txt:11 (find_package)
> 
> 
> -- Found Boost: D:/vcpkg-master/installed/x64-windows/include (found version "1.78.0") found components: filesystem system
> -- Performing Test COMPILER_HAS_DEPRECATED_ATTR
> -- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
> -- Performing Test COMPILER_HAS_DEPRECATED
> -- Performing Test COMPILER_HAS_DEPRECATED - Success
> Git LFS is not available, skipping download of datasets/openeb/sample_plugin_recording.raw
> Git LFS is not available, skipping download of datasets/metavision_core/csv_zip_eth
> -- Found OpenGL: opengl32
> -- Found GLEW: D:/vcpkg-master/installed/x64-windows/share/glew/glew-config.cmake
> -- Configuring done
> -- Generating done
> -- Build files have been written to: D:/Prophesee/openeb-2.3.0/build

I would appreciate it if you can help me to find where is the problem.

laurentbristiel, 2022-01-24:
Hi,
In your logs i can see that you are using the version 2.9 of Pybind whereas we need the version 2.6.
The problem probably comes from the fact that you are not using the proper version of vcpkg.
The one we require is this one: https://github.com/microsoft/vcpkg/archive/refs/tags/2020.11-1.zip
Could you make sure you are using this one?

laurentbristiel, 2022-01-24:
well, my mistake. In fact with the last version of OpenEB, the vcpkg version to use is this one: https://github.com/microsoft/vcpkg/releases/tag/2021.05.12
Can you try it? 
(doc update in progress)

sirius112358, 2022-01-24:
Thanks for you reply. 
I tried to use the vcpkg veresion 2021.05.12 to re-install the third-party libraries.
Then I build and compile the openeb code with CMake and the new vcpkg, but still it didn't work. 
The logs are as follows. It seems there are problems durint compiling, but I don't know what it means.

By the way, can I find the C++ version of the [Blinking Lights detector?](https://docs.prophesee.ai/stable/metavision_sdk/tutorials/Blinking_lights_detector.html)
Actually what I want to is using this algorithm to detect some blinking targets from SilkEvCam camera stream. 
The camera works well in C++, so if I can find the C++ code that's also OK. 

> **D:\Prophesee\openeb\build>cmake -A x64 -DCMAKE_TOOLCHAIN_FILE=D:\Prophesee\openeb\cmake\toolchains\vcpkg.cmake -DVCPKG_DIRECTORY=D:\vcpkg ..**
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19043.
-- The C compiler identification is MSVC 19.30.30528.0
-- The CXX compiler identification is MSVC 19.30.30528.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Microsoft Visual Studio/2022/Preview/VC/Tools/MSVC/14.30.30528/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Microsoft Visual Studio/2022/Preview/VC/Tools/MSVC/14.30.30528/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building METAVISION_SDK_MODULES_OPEN modules : base;core;driver;ui
-- Building METAVISION_SDK_MODULES_PROFESSIONAL modules : analytics;calibration;cv;ml;3dview;cv3d
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Found LibUSB: D:/vcpkg/installed/x64-windows/include/libusb-1.0
-- Found Boost: D:/vcpkg/installed/x64-windows/include (found version "1.75.0") found components: program_options filesystem timer chrono thread date_time atomic
-- Found Python3: D:/anaconda/python.exe (found version "3.9.7") found components: Interpreter Development Development.Module Development.Embed
-- Performing Test HAS_MSVC_GL_LTCG
-- Performing Test HAS_MSVC_GL_LTCG - Success
-- Found pybind11: D:/vcpkg/installed/x64-windows/include (found version "2.6.0" )
-- Found OpenCV: D:/vcpkg/installed/x64-windows (found version "4.5.1") found components: core highgui imgproc videoio imgcodecs calib3d objdetect
-- Found GTest: D:/vcpkg/installed/x64-windows/share/GTest/GTestConfig.cmake (found version "1.10.0")
---------------------------------------------------------------------------------------------------------
|  Test data was not found. To run the tests, please download the test data adapted to your project:    |
|  - OpenEB: https://dataset.prophesee.ai/index.php/s/VfhepWI4MPVWh9o (500 Mb archive)                  |
|  - Metavision Professional: https://dataset.prophesee.ai/index.php/s/JJ5xStgd8GSUWFz (3 Gb archive)   |
---------------------------------------------------------------------------------------------------------

Git LFS is not available, skipping download of datasets/openeb/gen31_timer.raw
Git LFS is not available, skipping download of datasets/openeb/gen4_evt2_hand.raw
Git LFS is not available, skipping download of datasets/openeb/gen4_evt3_hand.raw
Git LFS is not available, skipping download of datasets/openeb/blinking_leds.raw
Git LFS is not available, skipping download of datasets/openeb/gen4_evt2_hand_cd.dat
Git LFS is not available, skipping download of datasets/openeb/core
-- Found Boost: D:/vcpkg/installed/x64-windows/include (found version "1.75.0") found components: filesystem system
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
Git LFS is not available, skipping download of datasets/openeb/sample_plugin_recording.raw
Git LFS is not available, skipping download of datasets/metavision_core/csv_zip_eth
-- Found OpenGL: opengl32
-- Found GLEW: D:/vcpkg/installed/x64-windows/share/glew/glew-config.cmake
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Prophesee/openeb/build

**D:\Prophesee\openeb\build>cmake --build . --config Release --parallel 4**
Microsoft (R) Build Engine version 17.0.0-preview-21453-03+74e9935a4 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule D:/Prophesee/openeb/utils/cpp/pybind/lib/CMakeLists.txt
  Generating version file for library metavision_hal
  Generating version file for library metavision_sdk
[... 1166 lines cut ...]
D:\vcpkg\installed\x64-windows\include\gtest/gtest-matchers.h(533): message : see declaration of 'testing::internal::Ma
tcherBase<std::string>' [D:\Prophesee\openeb\build\sdk\modules\core\cpp\tests\gtest_metavision_sdk_core.vcxproj]
  threaded_process_gtest.cpp
D:\vcpkg\installed\x64-windows\include\gtest/gtest-matchers.h(511,56): warning C4275: non dll-interface class 'testing:
:MatcherDescriberInterface' used as base for dll-interface class 'testing::internal::MatcherBase<const std::string &>'
[D:\Prophesee\openeb\build\sdk\modules\core\cpp\tests\gtest_metavision_sdk_core.vcxproj]
[... cut ...]
tcherBase<std::string>' [D:\Prophesee\openeb\build\sdk\modules\core\cpp\tests\gtest_metavision_sdk_core.vcxproj]
D:\Prophesee\openeb\sdk\modules\core\cpp\tests\threaded_process_gtest.cpp(233,10): warning C4101: 'tester': unreference
d local variable [D:\Prophesee\openeb\build\sdk\modules\core\cpp\tests\gtest_metavision_sdk_core.vcxproj]
  Generating Code...
  gtest_metavision_sdk_core.vcxproj -> D:\Prophesee\openeb\build\bin\Release\gtest_metavision_sdk_core.exe
  Building Custom Rule D:/Prophesee/openeb/CMakeLists.txt

D:\Prophesee\openeb\build>

laurentbristiel, 2022-01-24:
> Then I build and compile the openeb code with CMake and the new vcpkg, but still it didn't work.
> The logs are as follows. It seems there are problems durint compiling, but I don't know what it means.

there are some warnings indeed (that I don't reproduce) during the first phase (creation of the makefiles with "cmake ..") but then the compilation looks OK. After this compilation, do you confirm that your build/py3 folder is still empty?

> By the way, can I find the C++ version of the Blinking Lights detector?

sorry but we only provide this tutorial in Python.

closed 2022-06-23 by laurentbristiel