prophesee-ai/openeb #40: How to tell if a raw file is in evt2 or evt3?

youkaichao, 2022-01-30:
Hi, I find that there are two decoders, one for evt2 and one for evt3.

Does the raw file contain the version information?

laurentbristiel, 2022-01-30:
Hello,

RAW file contain an ASCII header that has an evt2 or evt3 key in it.
See [Raw format documentation](https://docs.prophesee.ai/stable/data_formats/file_formats/raw.html) in our doc.

You can also use [metavision_raw_info application](https://docs.prophesee.ai/stable/metavision_sdk/modules/driver/guides/raw_info.html#chapter-sdk-driver-samples-raw-info) that will return the EVT2/EVT3 info as well. For example:

```
====================================================================================================

Name                80_balls.raw
Path                /path/to/80_balls.raw
Duration            6s 284ms 62us
Integrator          Prophesee
Plugin name         hal_plugin_gen3_fx3
Event encoding      EVT2  <============ what you are looking for
Camera generation   3.0
Camera systemID     21
Camera serial       0000ffff

====================================================================================================

Type of event       Number of events    First timestamp     Last timestamp      Average event rate
----------------------------------------------------------------------------------------------------
CD                  4588809             54                  6284062             730.2 Kev/s
```

Hope this helps!

closed 2022-01-30 by laurentbristiel

youkaichao, 2022-01-30:
Got it, thanks :)