prophesee-ai/openeb #43: Building Debian packages on Debian systems doesn't work

RalfGoebel, 2022-02-15:
Hi,

We are building the OpenEB Debian packages for Debian 10 and 11 systems, because our embedded devices do not support Ubuntu.
The CMake target _public_deb_packages_ used to work with OpenEB 2.2, but not anymore.

I think the problem is in python3.cmake, line 124:
https://github.com/prophesee-ai/openeb/blob/92aa7c1fc0df52ab53a0b62517255737689f1134/cmake/custom_functions/python3.cmake#L124

To fix the problem, I have replaced the line with:
`list(GET PYBIND11_PYTHON_VERSIONS 0 PYTHON3_DEFAULT_VERSION)`

I also added "3.9" to the variable _PYTHON3_ALL_VERSIONS_ in line 119 to make it work with Debian 11:
https://github.com/prophesee-ai/openeb/blob/92aa7c1fc0df52ab53a0b62517255737689f1134/cmake/custom_functions/python3.cmake#L119

Thanks,
Ralf

laurentbristiel, 2022-04-01:
Hi Ralf,
(belated) thanks for your feedback.
Your fix is legit and might help other people trying to compile OpenEB on Debiant 10/11.
At the moment we can not include this change in OpenEB because we don't have the proper CI env/tools to test another version of Python, but we will try to improve this in the future.
Laurent

closed 2022-04-01 by laurentbristiel

biphasic, 2022-09-15:
@RalfGoebel thanks for posting this. I'm also tied to Debian