prophesee-ai/openeb #45: ModuleNotFoundError: No module named 'metavision_hal'

jiangwei221, 2022-05-02:
Hi, I'm trying to run `metavision_raw_to_csv.py`, and compiled openeb as instructed, and ran the `setup_env.sh`.
When I ran `metavision_raw_to_csv.py`, I got:
```
Traceback (most recent call last):
  File "metavision_raw_to_csv.py", line 15, in <module>
    from metavision_core.event_io import EventsIterator
  File "/openeb/sdk/modules/core/python/pypkg/metavision_core/event_io/__init__.py", line 12, in <module>
    from .raw_reader import RawReader
  File "/openeb/sdk/modules/core/python/pypkg/metavision_core/event_io/raw_reader.py", line 22, in <module>
    from metavision_hal import DeviceDiscovery, RawFileConfig
ModuleNotFoundError: No module named 'metavision_hal'
```
Any ideas on how can I solve it?
Thanks!

laurentbristiel, 2022-05-02:
Hello @jiangwei221 
it could be because you are launching this Python script with a Python version we don't support.
We support Python 3.6 and 3.7 on Ubuntu 18 and Python 3.7 and 3.8 on Ubuntu 20.
What version of Python do you use?

biphasic, 2022-11-12:
@jiangwei221 you might want to have a look at our python data loader that we released https://github.com/fabhertz95/expelliarmus

biphasic, 2022-11-12:
it works on Python 3.7-3.11 on Windows, MacOs and Linux