prophesee-ai/openeb #48: Transform event data from ROS bag to HDF5 file

mgaoling, 2022-06-02:
Hi. I recently recorded a series of event data under ROS using the [prophesee_ros_wrapper](https://github.com/prophesee-ai/prophesee_ros_wrapper). This choice is because ROS provides a versatile framework that supports multi-sensors recording. However, I need to transform events from ROS bags to HDF5 files for now, as the ROS bag is quite bulky. Is there any function implemented in the metavision, or sample script to do so?

Thanks in advance!

laurentbristiel, 2022-06-03:
Hello LingGao
we don't have functions in Metavision to export ROS bags in other formats.
Do you have a specific HDF5 format in mind? Can you share what you would then do with those HDF5 files?
Laurent

mgaoling, 2022-06-03:
Thanks for the quick response.

As I mentioned above, to support data synchronization among different sensing modalities, we chose ROS as a middleware to record a few data sequences. However, some of my lab mates require events in HDF5 files as it is more popular among the python/learning community. [Format](https://dsec.ifi.uzh.ch/data-format/), such as the one used in DSEC and TUM-VIE dataset, with keys {events/x,y,p,t} and {ms_to_idx}, will be preferred.

mgaoling, 2022-06-03:
Thank you @nlyubova , this script is very helpful!

closed 2022-06-03 by mgaoling

nlyubova, 2022-06-03:
thanks to @f-zhen as she has proposed it ;)

biphasic, 2022-10-30:
for anyone else looking into this, may I also mention https://github.com/event-driven-robotics/importRosbag. The hdf5 bit has to be done manually though