prophesee-ai/openeb #50: Configure camera options (sensor pixel settings) using a python script?

andife, 2022-06-15:
I wonder if it is possible via Python script to use the following functionalities available via the Metavision Studio.

* Control sensor pixels settings (biases)
* Set a Region Of Interest (ROI)
* Access advanced sensor features (Anti-Flicker, Spatio-Temporal Contrast Filter and Event Rate Controller)

This functionality would be very interesting for parameter studies.

laurentbristiel, 2022-06-15:
Hello @andife 

yes, this is possible with the Python bindings of the HAL module.
See this documentation that contains some small examples: https://docs.prophesee.ai/stable/metavision_sdk/modules/hal/python_api/bindings.html

If you need larger example, there are some in our [Knowledge Center](https://support.prophesee.ai/portal/en/home) that you can access [if you have an account.](https://www.prophesee.ai/kc-access-request/).

Hope this helps,
Laurent

andife, 2022-06-15:
Hi Laurent,
thank you for the fast answer. I found what I need in the Knowledge Center. Thank you!

closed 2022-06-16 by laurentbristiel