prophesee-ai/openeb #55: Couldn't import `metavision_sdk_cv`

jjjkkkjjj-mizuno, 2022-06-30:
I couldn't import `metavision_sdk_cv` on Ubuntu22.04 with conda python=3.8.
I know my ubuntu version is not officially supported.
But i'd like to use openeb on ubuntu22.04 as possible as i can.
Do you have any advice?

Here is cmake generation output.

```bash
$ cmake .. -DBUILD_TESTING=OFF -DPython3_EXECUTABLE=/home/user/miniconda3/envs/evs/bin/python -DPYTHON3_DEFAULT_VERSION=3.8

-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building METAVISION_SDK_MODULES_OPEN modules : base;core;driver;core_ml;ui
-- Building METAVISION_SDK_MODULES_ADVANCED modules : analytics;calibration;cv;ml;3dview;cv3d
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.25
-- Found LibUSB: /usr/include/libusb-1.0  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: program_options filesystem timer chrono thread 
-- Found Python3: /home/user/miniconda3/envs/evs/bin/python (found version "3.8.13") found components: Interpreter Development Development.Module Development.Embed 
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/local/include (found version "2.6.0" )
-- Found OpenCV: /usr (found version "4.5.4") found components: core highgui imgproc videoio imgcodecs calib3d objdetect 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: filesystem system 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Generating internal bias files.
--  - vdac18_8m1.calib
--  - gen3_idac.calib
--  - gen3_idac_n_thick.calib
--  - gen3_idac_n_thin.calib
--  - gen3_idac_p_thick.calib
--  - gen3_idac_p_thin.calib
--  - gen3_idac_railp.calib
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.2.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/Documents/programs/openeb/build

```

Here is `sudo cmake --build . --target install` output.

```bash
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/metavision/licensing/LICENSE_OPEN
-- Installing: /usr/local/share/metavision/log/metavision-open-lib/meta-package.log
-- Installing: /usr/local/share/metavision/log/metavision-open-bin/meta-package.log
[... 485 lines cut ...]
-- Installing: /usr/local/share/cmake/MetavisionSDK/Modules/ui/MetavisionSDK_uiTargets-release.cmake
-- Installing: /usr/local/share/cmake/MetavisionSDK/Modules/ui/MetavisionSDK_uiConfig.cmake
-- Installing: /usr/local/share/cmake/MetavisionSDK/Modules/ui/MetavisionSDK_uiConfigVersion.cmake
-- Installing: /usr/local/share/cmake/MetavisionSDK/Modules/ui/MetavisionSDK_uiDepends.cmake
-- Up-to-date: /usr/local/share/metavision/sdk/ui/samples/metavision_simple_window/metavision_simple_window.cpp
-- Up-to-date: /usr/local/share/metavision/sdk/ui/samples/metavision_simple_window/constants.cpp.in
-- Up-to-date: /usr/local/share/metavision/sdk/ui/samples/metavision_simple_window/constants.h
-- Up-to-date: /usr/local/share/metavision/sdk/ui/samples/metavision_simple_window/prophesee_logo.jpg
-- Up-to-date: /usr/local/share/metavision/sdk/ui/samples/metavision_simple_window/CMakeLists.txt
-- Installing: /home/user/miniconda3/envs/evs/lib/python3.8/site-packages/metavision_sdk_ui.cpython-38-x86_64-linux-gnu.so
-- Set runtime path of "/home/user/miniconda3/envs/evs/lib/python3.8/site-packages/metavision_sdk_ui.cpython-38-x86_64-linux-gnu.so" to ""
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_decoder
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_decoder/README.md
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_decoder/CMakeLists.txt
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_decoder/metavision_evt2_raw_file_decoder.cpp
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt3_raw_file_decoder
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt3_raw_file_decoder/metavision_evt3_raw_file_decoder.cpp
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt3_raw_file_decoder/README.md
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt3_raw_file_decoder/CMakeLists.txt
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_encoder
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_encoder/README.md
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_encoder/CMakeLists.txt
-- Up-to-date: /usr/local/share/metavision/samples/metavision_evt2_raw_file_encoder/metavision_evt2_raw_file_encoder.cpp

```

laurentbristiel, 2022-07-04:
Hello @jjjkkkjjj-mizuno 
The module `CV` is not part of OpenEB.
This why you can't import `metavision_sdk_cv`.

You can see some explanations in that doc page: https://docs.prophesee.ai/stable/modules.html
You will see that only the "fundamental/pillar" modules are part of OpenEB.
They are enough to have make your camera work (start, stream..) and to manipulate data (decode, iterate on events..).
But you don't have the whole set of algo (like those in CV module).

We plan to support Ubutun 22 in a future release.
If you want the "full Metavision Intelligence source code", it is possible to contact sales@prophesee.ai to have a quote.
Or maybe share here what you would like to do with CV module and we could find workarounds.

jjjkkkjjj-mizuno, 2022-07-06:
I'm sorry for not reading the documents and thank you for sharing proper information.
I understood openeb is fundamental module so it has no "advanced" CV module.

Officially supporting Ubuntu 22 is good news for me.

I'd like to create the app to reconstruct 3D coordinates from 2 event-based cameras using [Sync sample](https://docs.prophesee.ai/stable/metavision_sdk/modules/core/guides/synchronization.html#chapter-samples-sync).
I thought it is better option to use openeb than SDK to achieve this project.
But this project is just getting started, and project's progress level is trying the sample code first. (In the meantime, I tried to use CV module and get the error.)
Therefore, I'll use some sample codes next and ask you (or contact sales) when I want workarounds for "advanced" module.

Again, thank you for your advice.

closed 2022-07-06 by jjjkkkjjj-mizuno