prophesee-ai/openeb #59: Missing RawReader.duration_s (or document mismatch)

shiba24, 2022-10-06:
Hello team,

The official document says that `RawReader` has the `duration_s` attribute: https://docs.prophesee.ai/stable/metavision_sdk/modules/core/python_api/event_io.html#metavision_core.event_io.RawReader.duration_s

However, it seems that it does not actually. Only this comment exists and I cannot find any code: https://github.com/prophesee-ai/openeb/blob/main/sdk/modules/core/python/pypkg/metavision_core/event_io/raw_reader.py#L381

Is the documentation correct?

lbristiel-psee, 2022-10-07:
Hello @shiba24 ,
this is indeed a documentation issue.
This attribute is not available in fact.
We will fix the doc for next release, and see if this class should be reviewed to have such a parameter available.
Thanks for your feedback.

shiba24, 2022-10-07:
Understood, thank you. This `duration_s` and the number of events in total for `RawReader` or `EventsIterator` are very useful from a user's perspective, so I hope you guys will support them in future. Closing the issue.

closed 2022-10-07 by shiba24