prophesee-ai/openeb #73: Reading RAW files without having to install everything?

agirbau, 2022-12-08:
Hello, 

I was wondering if there is a way to read a raw file in Python without having to install all the SDK. Thanks a lot!

lbristiel-psee, 2022-12-08:
Hello @agirbau ,

Currently there is no sub-part/module of OpenEB that you could install to have access to this feature only (read raw file in Python). If compiling OpenEB is a problem for you, note that you can install a pre-built binary version (https://www.prophesee.ai/metavision-intelligence-sdk-download/) that is not so long/complicated to install (see https://docs.prophesee.ai/stable/installation/linux.html)

Though, if you want "Read RAW file" feature only, you can take a look at this repo: https://github.com/fabhertz95/expelliarmus on which @biphasic and @fabhertz95 are working.

fabhertz95, 2022-12-08:
Thanks @lbristiel-psee :) Yes @agirbau, we support DAT, EVT2 and EVT3 encodings, with pip install available.

closed 2022-12-09 by agirbau