prophesee-ai/openeb #75: Question to check data quality

shiba24, 2023-02-09:
Hi, I have a question to record better data.
Is there a way (an API) to check if the event data is in good quality during recording?

My concern is when the events are too many during recording, such as too fast scenes, lighting, etc.
Some other cameras like DAVIS etc may suffer from event data drop or the saturation of timestamps as you know.
I imagine there are similar limitations for Prophesee cameras, too. (Please let me know if this assumption is wrong.)

Is there a way to check
- if any event data are dropped (randomly) due to the limitation of bandwidth or max MEv/s?
- the possibility of the saturation of timestamps - like many events are assigned almost the same quantized timestamps
Or any other quality deterioration that you have been aware of?

Just getting the information about if the data is good or not helps us to record data with appropriate settings or re-record it immediately.

Thank you!

lbristiel-psee, 2023-02-24:
Hello,

sorry for the delay to answer your interesting question.

So first, your assumption is correct: when the sensor is facing a very dynamic scene or moving in front of a rich scene, some artefacts may appear when viewing the events frame by frame. This can be seen in the image below, where some horizontal lines without events are visible. 

[image]

The explanation is that the number of events that the sensor can communicate through its interface is bounded. In some contexts where many events are captured, we can reach the limit of what the interface can handle. We call that phenomenon "readout saturation". In the image above, this explains the "empty lines". Essentially, there should be events in those location, but they were lost.

In the current generation of sensors, there is no data/information provided that would inform about such a saturation. What we advise is to do some manual testing with your setup and start by visually checking (in Visual Studio for example) if the shown frames look OK (no artifact) and until what even rate it remains OK. Then, if you have a Gen4 sensor (Gen41, IMX636) you can set an [ERC level](https://docs.prophesee.ai/stable/hw/manuals/esp.html#event-rate-controller-erc) to limit the bandwidth of the events. 

This being said, the best way is to avoid readout saturation in the first place. To do so, you need to reduce the number of events captured by the sensor. This can be done in several ways summarized in [Metavision Studio documentation](https://docs.prophesee.ai/stable/metavision_studio/index.html#first-recording-from-live-camera):

- Camera installation: when possible, mount your camera on a tripod stand to avoid any spurious motion during acquisition
- Lighting conditions: like mentioned in the previous section, make sure you don’t have unexpected events due to flickering light source.
- Focus adjustment: to help you focusing your camera, you can use the [metavision_blinking_pattern_focus application](https://docs.prophesee.ai/stable/metavision_sdk/modules/calibration/samples/focus.html#chapter-sdk-calibration-samples-focus)
- Sensor pixels settings (biases): depending on your applications requirements and conditions (higher speed, lower background activity, higher contrast sensitivity threshold, etc.), you should adjust the [camera biases](https://docs.prophesee.ai/stable/hw/manuals/biases.html#chapter-biases).
- Region Of Interest: whenever possible you should limit the area of the sensor to the pixels that might gather relevant events for your application. For example, if you are tracking vehicle on a road and you are unable to adjust your objective lens to see only the road, you can configure a ROI in Metavision Studio to exclude non-relevant areas (sidewalks, sky etc.).
- Events filtering: if your camera is using a Gen4 or IMX636 sensor, leverage the Event Signal Processing (ESP) block that provide some event filtering features: [Anti-Flicker](https://docs.prophesee.ai/stable/hw/manuals/esp.html#chapter-esp-afk), [Event Trail Filter](https://docs.prophesee.ai/stable/hw/manuals/esp.html#chapter-esp-stc-trail) and [Event Rate Controller (ERC)](https://docs.prophesee.ai/stable/hw/manuals/esp.html#chapter-esp-erc).

Hope this helps,
Laurent

shiba24, 2023-02-24:
Hi @lbristiel-psee , thank you always for your support. This question is not urgent so please take your time.
Also, thank you for the sample data. It is very interesting to see these "empty lines", especially that they exist only in a part of the image plane (in this example, bottom center of the image plane). I thought that if this kind of data drop happens, they would be everywhere in the image plane. Do you know why?

I remember that in some previous experimental recordings that we did for fast and complex motion, we get similar readout saturation. In the end we changed the refractory period and set ROI as a workaround (so, it is not a problem for now).

For your information, here is some analysis that I did.
In the figure below, the horizontal axis is time in seconds (so the entire range is for 1 ms), and the vertical axis is the number of event in the entire image plane, aggregated in each micro-second window. (i.e., event rate with 1000 bins for 1 ms; 20-ish events/microsec = 20-ish MEv/s).
You see there are some sudden "zero" event moments (where bins are missing) starting after x=7.0004 sec. I think this is what you call readout saturation. In our case it was in the entire image plane, not only a part.
This event-missing was about in every 50 micro second in my impression, if I remember correctly (but probably depends on refr).

[image]

I cannot tell if the cause of these saturation is actually dropped events (so I can trust the quality of events (especially timestamps) for the rest of the data), or the saturation or mal-assignment of timestamps (events are recorded but the timestamp might be not as accurate as other part of data).

From our side, it would be great that if you could detect these readout saturation programatically and expose this information to users (I understand it is not easy).
This is because metavision SDK has APIs to adjust these parameters. So users might be able to explore and write their own program how to adjust these parameters dynamicly or interactively to improve the recording quality.

Shintaro

lbristiel-psee, 2023-03-04:
> I thought that if this kind of data drop happens, they would be everywhere in the image plane. Do you know why?

Keep in mind that the image above is an accumulation of events during a certain duration (for example 20ms) so the notion of "saturation over the whole image plane" does not really apply. To see what really happened, one would have to check microsecond by microsecond (which you also did, see next questions). And alo, note that in Prophesee event-based sensor, events are captured line by line. So for this image, maybe some portions of the images during those 20ms did not cause any saturation, but at a given moment, there were some portion of the image plan that received too much events in a very short duration which caused saturation and errors in the events (either loss or incorrect timestamp). This is why the drop can be "local".

> I remember that in some previous experimental recordings that we did for fast and complex motion, we get similar readout saturation. 
> In the end we changed the refractory period and set ROI as a workaround (so, it is not a problem for now).

This is exactly what you should do: adjusting biases and ROI.
You can also adjust your lightning, the camera position/movement etc as mentioned in my previous answer.

> For your information, here is some analysis that I did.

could you share with what sensor/evk you did those experiments?

> You see there are some sudden "zero" event moments (where bins are missing) starting after x=7.0004 sec. 
> I think this is what you call readout saturation. 

Well, readout saturation can lead to drop of events, but most of the time the consequence would be a loss of precision on the timestamping. The ideas is that when the throughput capability of the sensor readout pipeline and IO interface is exceeded continuously, it can exert a “backpressure” up to the pixel array when all the events cannot be read at the same time and influence the timing precision of single events.

In your case, those "zero moment" are for a single micro-second right? It does not mean that some events were lost or timeshifted, it could mean that the sensor just did not detect any event during this micro-second. Your event rate is below what our sensor can handle, so I doubt you are facing saturation here. 

> From our side, it would be great that if you could detect these readout saturation programatically and expose this information to users

yes, I understand your request and it makes sense. I will circulate it around at Prophesee.

In addition, if you are a Prophesee customer (if you own an EVK), I would suggest to request an account in our Knowledge Center (https://www.prophesee.ai/kc-access-request/) so that you can access Application Note about readout of our sensors that could help you understand the problem you are facing.

Laurent for Prophesee Support

shiba24, 2023-03-13:
> in Prophesee event-based sensor, events are captured line by line. So for this image, maybe some portions of the images during those 20ms did not cause any saturation, but at a given moment, there [...]

This makes sense and explains the locality of event drop. Thanks!

> could you share with what sensor/evk you did those experiments?

It's EVK3 Gen4.1.

> most of the time the consequence would be a loss of precision on the timestamping

I see. This question is now more of a curiosity, what is the precision on timestamps for Psee cams that we users can assume, do you think?
Basically, event-vision researchers write the sentences like "event cameras have precise temporal resolution on the order of micro sec." But practically for the current Psee sensors, considering the "loss of precision on the timestamping", the timestamp precision varies between scenes, not only based on the hardware itself, sometimes actual microseconds and sometimes 10-microseconds? (Well in most current research papers, even 10-micro or 100-micro sec may be good enough, but I would like to know the limitation.)

> It does not mean that some events were lost or timeshifted, it could mean that the sensor just did not detect any event during this micro-second. 

But this is not likely to me. The scene that we recorded includes very continuous physical motion in the scene so I don't think there were such actual no-event moments. I will be able to share the recording settings when our under-review paper is published.

Yes, I'm a Prophesee customer, I'm working under Guillermo (TU Berlin). I will check the knowledge center too.

lbristiel-psee, 2023-04-18:
Hi @shiba24 
Sorry for the delay to come back to you.
So, the 1us precision on timestamp is really the granularity in which events are timestamped, but like you mention the actual precision we will get depends on the scene. I am unable to give you numbers, but as you confirmed you are a Prophesee customer, I would recommend that you read our [Knowledge Center Page "Temporal precision of Gen4 sensor's readout"](https://support.prophesee.ai/portal/en/kb/articles/temporal-precision-of-gen4-sensor-s-readout ) (if you did not read it yet). Understanding more in details the readout process could help you better grasp the limitation and also see how you can push the sensor to its limit.
BR,
Laurent for Prophesee Support

closed 2023-06-09 by lbristiel-psee