prophesee-ai/openeb #79: OpenEB 4.0 index fails to build

E1k3, 2023-04-13:
After upgrading to 4.0, some but not all existing RAW files fail to be read depending on the application used:

`metavision_file_to_hdf5` repeatedly fails to build the index, but `metavision_file_info` succeeds at building the index and afterwards, every sample is able to load the index cache just fine.

```
$ metavision_file_to_hdf5 -i 2022-10-04_15-36-54.raw
Wrote HDF5 file  2022-10-04_15-36-54.hdf5
[HAL][ERROR] Failed to build index for input RAW file 2022-10-04_15-36-54.raw

$ metavision_file_to_hdf5 -i 2022-10-04_15-36-54.raw
Wrote HDF5 file  2022-10-04_15-36-54.hdf5
[HAL][ERROR] Failed to build index for input RAW file 2022-10-04_15-36-54.raw

$ metavision_file_info -i 2022-10-04_15-36-54.raw
====================================================================================================

Name                2022-10-04_15-36-54.raw
Path                /redacted/2022-10-04_15-36-54.raw
Duration            1m 49s 595ms 996us
Integrator          Prophesee
Plugin name         hal_plugin_gen41_evk3
Data encoding       EVT3
Camera generation   4.1
Camera systemID     48
Camera serial       00042340

====================================================================================================

Type of event       Number of events    First timestamp     Last timestamp      Average event rate
----------------------------------------------------------------------------------------------------
CD                  20499249            817                 109595996           187.0 Kev/s
External triggers   21063               2227111             107159923           192 ev/s

$ metavision_file_to_hdf5 -i 2022-10-04_15-36-54.raw
Wrote HDF5 file  2022-10-04_15-36-54.hdf5
```

Our application uses `I_EventsStreamDecoder` and behaves like `metavision_file_to_hdf5` as in it crashes for some, but not all files that do not already have an index cache generated by `metavision_file_info`.

On a related note, it would be nice to make RAW file indexing in `EventsStream` optional, because we only read events sequentially and do not seek.

lbristiel-psee, 2023-04-16:
Hello,

> After upgrading to 4.0, some but not all existing RAW files fail to be read depending on the application used:
> `metavision_file_to_hdf5` repeatedly fails to build the index, but [...]

Indeed, we were able to reproduce the problem which is a change from 3.x to 4.0
But note that the error you get on the index building ([HAL][ERROR] Failed to build index) is in fact only an information that the index of the RAW file was not build, it does not prevent the rest of the program to work properly. In the case of metavision_file_to_hdf5 for example, the HDF5 file should be created even if you get the error.

For information, the reason why there is an index error is because the program (here metavision_file_to_hdf5) goes very fast and the index that is built in background has no time be completed, so we get an error.

> On a related note, it would be nice to make RAW file indexing in `EventsStream` optional, because we only read events sequentially and do not seek.

This is in fact possible, though not yet properly documented.
The idea is to use Metavision::FileConfigHints() that can configure multiple options for file reading.
To disable the RAW file indexing, the statement is:

```
    camera_file_conf.set("index", false);
```

And so, if you want to modify metavision_file_to_hdf5 (and/or your code) to not build the index file (which is optional anyway, only used to seek in the file), you should replace this statement:

```
    camera = Metavision::Camera::from_file(in_file_path, Metavision::FileConfigHints().real_time_playback(false));`
```
		
by those ones:

```
	Metavision::FileConfigHints camera_file_conf;
        camera_file_conf.real_time_playback(false);
        camera_file_conf.set("index", false);
        camera = Metavision::Camera::from_file(in_file_path, camera_file_conf);
```
		
On our side, we are going to update the metavision_file_to_hdf5 this way for next release.
We will also enhance the documentation to be clearer about this configuration.
Finally we will lower the severity of the "Failed to build index" as the "error" level can be misleading.

lbristiel-psee, 2023-04-21:
Hello @E1k3 
Hpe you wre able to fix your issue.
On our side we did some modifications in the source code after you raised this problem.
Now the ERROR is a WARNING and is not shown when the index built is aborted.
See this commit: https://github.com/prophesee-ai/openeb/commit/09c57ca3a58c99ad515c548289eec6be9b4ae895#diff-1f87c0408024b5f36c106f0ca65aae0917993ae3144be26e1feb2aef1be1030f
This is available in OpenEB 4.0.1that was just released.

And the doc was also updated: 
https://docs.prophesee.ai/stable/data/file_formats/raw.html#raw-index-file

If this is now OK for you, can we close this issue?

closed 2023-04-21 by lbristiel-psee