prophesee-ai/openeb #80: How to avoid opencv and gtk being linked to code

ivangalysh, 2023-04-13:
I just downloaded the new openeb and recompiled my code that grabs events, reformats and passes the results to another program. It has been working with the older version of openeb. The newly compiled code is looking for opencv libraries and gtk libraries.  How do I stop getting these libraries linked in when I don't use them. I am modifying the metavision_sdk_get_started.cpp

I have been cross compiling the code to operate on an ARM processor. I would like to use the new openeb since it has added the pixel masking.

Thanks
Ivan

ogeorget-psee, 2023-04-14:
Hi @ivangalysh, thanks for your feedback !

Could you please share the exact error you are getting, so we can better understand your issue ? 

If you want to modify the linked libraries, you can have a look at the CMakeList.txt for the related app : https://github.com/prophesee-ai/openeb/blob/ea2cf58a636a6df85c607adc1c857d47c525474b/sdk/modules/core/cpp/samples/metavision_sdk_get_started/CMakeLists.txt#L11.

Getting rid of the dependency on `opencv_highgui` should also remove `gtk` as it's one of its dependencies.

ivangalysh, 2023-04-17:
Sorry for the late response, I was on travel.

The error message when I run my program is:
./camera_h3: error while loading shared libraries: libopencv_highgui.so.4.5: cannot open shared object file: No such file or directory

I did remove opencv_highgui from the CmakeLIst.txt. I even removed it from the CmakeList.txt.install file.

Ivan

lbristiel-psee, 2023-04-21:
Thanks for the feedback @ivangalysh 
There was indeed an error in SDK Get Started sample which was wrongly requiring OpenCV in CMake file.
We fixed this in Release 4.0.1
See this commit: https://github.com/prophesee-ai/openeb/commit/09c57ca3a58c99ad515c548289eec6be9b4ae895#diff-029d0f0902bede65de5f460c0797cb1359579a5261018b6e09fe30b3bf5545f2

Is situation OK for you now? 
If so, can we close this issue?

ivangalysh, 2023-04-21:
Here is the result of ldd on the program. It appears to want a large number of libraries.
I recompiled the older version to see if I get the same result and it doesn't link in everything.

I tried this on two other computers and get the same results.

Ivan

galysh@galysh-virtual-machine:~/openeb/build$ ldd bin/metavision_sdk_get_started 
	linux-vdso.so.1 (0x00007ffcdaa67000)
	libmetavision_sdk_driver.so.4 => /home/galysh/openeb/build/lib/libmetavision_sdk_driver.so.4 (0x00007f9dd1374000)
	libmetavision_sdk_ui.so.4 => /home/galysh/openeb/build/lib/libmetavision_sdk_ui.so.4 (0x00007f9dd1355000)
	libmetavision_sdk_core.so.4 => /home/galysh/openeb/build/lib/libmetavision_sdk_core.so.4 (0x00007f9dd130e000)
	libopencv_core.so.4.5d => /lib/x86_64-linux-gnu/libopencv_core.so.4.5d (0x00007f9dd0e00000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9dd0a00000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9dd12d3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9dd0600000)
	libmetavision_hal_discovery.so.4 => /home/galysh/openeb/build/lib/libmetavision_hal_discovery.so.4 (0x00007f9dd12b2000)
	libboost_filesystem.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0 (0x00007f9dd1292000)
	libhdf5_serial_cpp.so.103 => /lib/x86_64-linux-gnu/libhdf5_serial_cpp.so.103 (0x00007f9dd121b000)
	libhdf5_serial.so.103 => /lib/x86_64-linux-gnu/libhdf5_serial.so.103 (0x00007f9dd0200000)
	libhdf5_ecf_codec.so.1 => /home/galysh/openeb/build/lib/libhdf5_ecf_codec.so.1 (0x00007f9dd1211000)
	libmetavision_hal.so.4 => /home/galysh/openeb/build/lib/libmetavision_hal.so.4 (0x00007f9dd11bb000)
	libmetavision_sdk_base.so.4 => /home/galysh/openeb/build/lib/libmetavision_sdk_base.so.4 (0x00007f9dd11a6000)
	libglfw.so.3 => /lib/x86_64-linux-gnu/libglfw.so.3 (0x00007f9dd0d99000)
	libGLEW.so.2.2 => /lib/x86_64-linux-gnu/libGLEW.so.2.2 (0x00007f9dd0cdb000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f9dd1170000)
	libOpenGL.so.0 => /lib/x86_64-linux-gnu/libOpenGL.so.0 (0x00007f9dd0caf000)
	libopencv_highgui.so.4.5d => /lib/x86_64-linux-gnu/libopencv_highgui.so.4.5d (0x00007f9dd0c7d000)
	libopencv_videoio.so.4.5d => /lib/x86_64-linux-gnu/libopencv_videoio.so.4.5d (0x00007f9dd097a000)
	libopencv_imgproc.so.4.5d => /lib/x86_64-linux-gnu/libopencv_imgproc.so.4.5d (0x00007f9dcfc00000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9dd0893000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f9dd0c5f000)
	libtbb.so.2 => /lib/x86_64-linux-gnu/libtbb.so.2 (0x00007f9dd084d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f9dd1452000)
	libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x00007f9dcf600000)
	libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x00007f9dd0159000)
	libsz.so.2 => /lib/x86_64-linux-gnu/libsz.so.2 (0x00007f9dd0c5a000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f9dcfac0000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f9dcf579000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f9dcf4c1000)
	libopencv_imgcodecs.so.4.5d => /lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.5d (0x00007f9dd05a7000)
	libgtk-3.so.0 => /lib/x86_64-linux-gnu/libgtk-3.so.0 (0x00007f9dcec00000)
	libgdk-3.so.0 => /lib/x86_64-linux-gnu/libgdk-3.so.0 (0x00007f9dceaf9000)
	libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f9dce9d1000)
	libgdk_pixbuf-2.0.so.0 => /lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f9dcfa90000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f9dcf461000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f9dce897000)
	libdc1394.so.25 => /lib/x86_64-linux-gnu/libdc1394.so.25 (0x00007f9dce81f000)
	libgstbase-1.0.so.0 => /lib/x86_64-linux-gnu/libgstbase-1.0.so.0 (0x00007f9dce79a000)
	libgstreamer-1.0.so.0 => /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 (0x00007f9dce649000)
	libgstapp-1.0.so.0 => /lib/x86_64-linux-gnu/libgstapp-1.0.so.0 (0x00007f9dd0c3e000)
	libgstriff-1.0.so.0 => /lib/x86_64-linux-gnu/libgstriff-1.0.so.0 (0x00007f9dd0c2d000)
	libgstpbutils-1.0.so.0 => /lib/x86_64-linux-gnu/libgstpbutils-1.0.so.0 (0x00007f9dcfa4c000)
	libgstvideo-1.0.so.0 => /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0 (0x00007f9dce584000)
	libavcodec.so.58 => /lib/x86_64-linux-gnu/libavcodec.so.58 (0x00007f9dcd000000)
	libavformat.so.58 => /lib/x86_64-linux-gnu/libavformat.so.58 (0x00007f9dccc00000)
	libavutil.so.56 => /lib/x86_64-linux-gnu/libavutil.so.56 (0x00007f9dcc800000)
	libswscale.so.5 => /lib/x86_64-linux-gnu/libswscale.so.5 (0x00007f9dce4e6000)
	libgphoto2.so.6 => /lib/x86_64-linux-gnu/libgphoto2.so.6 (0x00007f9dce461000)
	libgphoto2_port.so.12 => /lib/x86_64-linux-gnu/libgphoto2_port.so.12 (0x00007f9dd083f000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007f9dd012f000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x00007f9dcf440000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007f9dce442000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x00007f9dccf93000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x00007f9dd0829000)
	libssl.so.3 => /lib/x86_64-linux-gnu/libssl.so.3 (0x00007f9dcceef000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007f9dcce9b000)
	libldap-2.5.so.0 => /lib/x86_64-linux-gnu/libldap-2.5.so.0 (0x00007f9dccba1000)
	liblber-2.5.so.0 => /lib/x86_64-linux-gnu/liblber-2.5.so.0 (0x00007f9dd0594000)
[... cut ...]
	libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x00007f9dbf400000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007f9dc146f000)
	libgfortran.so.5 => /lib/x86_64-linux-gnu/libgfortran.so.5 (0x00007f9dbf000000)
	libnssutil3.so => /lib/x86_64-linux-gnu/libnssutil3.so (0x00007f9dc143d000)
	libplds4.so => /lib/x86_64-linux-gnu/libplds4.so (0x00007f9dc1812000)
	libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x00007f9dc1432000)
	libquadmath.so.0 => /lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f9dc11b8000)

ogeorget-psee, 2023-04-24:
Hi @ivangalysh 

The base library `metavision_sdk_core` also depends on `opencv_highgui` as you can see here. 
https://github.com/prophesee-ai/openeb/blob/09c57ca3a58c99ad515c548289eec6be9b4ae895/sdk/modules/core/cpp/lib/CMakeLists.txt#L29

This is a requirement for [simple_displayer.cpp](https://github.com/prophesee-ai/openeb/blob/main/sdk/modules/core/cpp/src/simple_displayer.cpp) that makes use of OpenCV Gui.

If you don't use it, you can comment it our from the CmakeList here :
https://github.com/prophesee-ai/openeb/blob/09c57ca3a58c99ad515c548289eec6be9b4ae895/sdk/modules/core/cpp/src/CMakeLists.txt#L21.

This should get rid of the dependency.

ivangalysh, 2023-04-24:
I did that and more to where I was able to generate the libraries with no references to opencv and gtk and my code also doesn't want those libraries.

Along with the simple_displayer.cpp, I had to comment out anything that generates a frame or does anything with video. The overall cmake fails after compiling and linking metavision_sdk_getting_started so the code and libraries I want do get compiled and linked.

I think I am good for now. I just have to learn how to add pixel masking to my code. Does pixel masking require opencv?

You can close this out.
Thanks for all your help.
Ivan

lbristiel-psee, 2023-04-25:
Hello @ivangalysh ,
we will clean our code to avoid unnecessary OpenCV dependency in our next version. Thanks for the feedback.
About your question on "pixel masking", can you explain what you are willing to achieve?
Best,
Laurent

ivangalysh, 2023-04-28:
Thanks for pointing me to this. I am now using the showcase code as my base. I was able to remove all GUI and opencv and it compiles properly and runs on the ARM computer. I am now customizing it to meet my requirements.

Thanks for your help.

Ivan

ivangalysh, 2023-04-28:
Quick question. How many pixels can I mask? I checked the size of the mask and it looks like up to 64 can be added. Is that correct?

Ivan

lbristiel-psee, 2023-04-30:
> Quick question. How many pixels can I mask? I checked the size of the mask and it looks like up to 64 can be added. Is that correct?

yes, this is correct, you can mask up to 64 pixels.

closed 2023-04-30 by lbristiel-psee