prophesee-ai/openeb #88: help with building a rosdistro package

berndpfrommer, 2023-07-05:
OpenEB team:
I'm currently working on building a [ROS metavision](https://github.com/ros-event-camera/metavision_driver) package for release via "rosdistro". This will require some of the openeb libraries to be shipped along with the ROS driver.
Is there somebody I could work with to answer a few questions:
1) Is it legal to ship the compiled binaries as part of a ROS package?
2) Are there any restrictions on downloading OpenEB to the ROS compile servers and compiling them there in an automated fashion?
Would be great to have one of your developers on the release team (ros-event-cameras). I invited @laurentbristiel but please suggest other team members if appropriate.
Thanks,
Bernd

lbristiel-psee, 2023-07-07:
Hello Bernd,
Thanks for the proposition. That's look interesting and promising!
We are discussing it internally to double-check that it works OK for us and come back to you shortly.
One question: what would be the roles (and skills needed) for the person that would be part of the release team of ros-event-cameras?
Thanks,
Laurent

berndpfrommer, 2023-07-08:
Laurent,
To answer your question: 
- first and foremost I'd like to have somebody from Prophesee with a github account become a member of the "ros-event-camera" organization. This is a github organization that provides basic ROS/ROS2 source packages for event camera support (ROS driver and encoding/decoding software). The person joining should have a basic grasp of Linux, git, github, and maybe ROS, although the latter is not too crucial. The important part is that they can look over our package (other members of the team will of course provide help if necessary) to determine that we do not accidentally violate any copyright / intellectual property rights of Prophesee. So whoever joins "ros-event-camera" should be able to interface with Prophesee's legal expert(s).
- I don't expect this to be a substantial workload going forward. I will be driving the maintenance and development of all packages, and it's really only the driver at the moment that is directly using the MetavisionSDK (and where creating a ROS package will require shipping Metavision library binaries along with it). I assume that once the ROS driver package has been released, very little involvement from your side would be required.
Regards,
Bernd

lbristiel-psee, 2023-07-13:
Hello Bernd,

sorry for the little delay to come back to you.
I wanted to circle around internally properly before coming back to you.
So about your various points:

0) first, thanks again for your interest in our technology and the effort you put into this ROS/EB ecosystem!

1) yes, it is legal to ship compiled binaries of OpenEB as part of ROS package.
Essentially, OpenEB is under Apache2 licence which allows to distribute object code. 

2) there is no restriction on downloading OpenEB to the ROS compile servers and compiling them there in an automated fashion.
Here again, with the Apache2 licence of OpenEB you are allowed to download the source code and compile it.

3) I can be Prophesee person that becomes a member of the "ros-event-camera" organization.
I am not a core software developer but I am rather in charge of Customer Experience working on support/documention/delivery topics. I hope I should be able to help you.

Best,
Laurent

lbristiel-psee, 2023-07-13:
@berndpfrommer you invited me through my acount @laurentbristiel which is my personal account. Could you invite me through my Prophesee account whith is @lbristiel-psee ?

berndpfrommer, 2023-07-13:
Done! Thanks for your help!

berndpfrommer, 2023-08-04:
The rosdistro package is building now, thanks for your support.
There will be some related pull requests against openeb coming forth.

closed 2023-08-04 by berndpfrommer