prophesee-ai/openeb #94: Questions regarding to support Prophesee cameras in PyAER

duguyue100, 2023-09-14:
Hi, all,

I'm the author of [PyAER](https://github.com/duguyue100/pyaer), which supports iniVation cameras.
I'm starting to look into expanding the support to Prophesee cameras.

Before I spend too much time researching how I could integrate PyAER, I would like to ask some basic questions:

1. I see the software is mainly tested on Ubuntu, have you tried to build it on Mac? Especially Mac with M-series chips?
2. Are there packaging plans such as pushing prebuilt libraries into a PPA repository and Homebrew?
3. I prefer that I don't have to install some of the heavy dependencies such as `libopencv`, `libboost`, and some of the visualization libraries. Is there a minimal component that I could build and use? Likely I only need the event readout, configuration, and filtering functions. (Looks like I should build HAL and Core)

Thanks for your support in advance! 

Regards,
Yuhuang.

lbristiel-psee, 2023-09-15:
Hello Yuhuang,

congrats for your job on PyAER and thanks for your interest in Prophesee sensors/cameras/SDK.
Let me try to help you on your questions.

> have you tried to build it on Mac?

There is no official support, but OpenEB was already compiled on MacOS.
Depending on the versions of OpenEB and MacOS, it requires some light modifications in the code, but nothing major.

> Especially Mac with M-series chips?

No, we did not test on M-series Mac.
Note though that compilation on Mac was discussed in this issue: https://github.com/prophesee-ai/openeb/issues/58
Maybe you will find some useful information in there.

> Are there packaging plans such as pushing prebuilt libraries into a PPA repository and Homebrew?

Not at the moment.

> Is there a minimal component that I could build and use? Likely I only need the event readout,
> configuration, and filtering functions. (Looks like I should build HAL and Core)

From OpenEB, HAL and Base modules and indeed required to decode/stream in Python. It should be possible to strip down OpenEB to not use libopencv and libboost if you don't include the apps/samples and keep only those 2 modules. Filtering is more in Core module but this one has boost and opencv dependency. 

Hope this helps,
Laurent for Prophesee Support

duguyue100, 2023-09-15:
@lbristiel-psee Awesome, thanks for the report, I will come back if I have any more questions, I will close this issue for now.

closed 2023-09-15 by duguyue100

dlzcn, 2023-09-18:
@duguyue100 OpenEB, including both C++ and pybinds works well on Apple-Silicon. As mentioned by Laurent, you may need to modify some codes in order to compile them on MAC.