prophesee-ai/openeb #98: Feature Request: Add resize method to EventCDBuffer in python

richbai90, 2023-11-16:
Right now, if you want to build custom processing stages in python it's a bit of a hassle because there isn't any standard way to do it. It could be facilitated easily by creating a contract that all filters in the pipeline need to follow the convention of the existing Metavision filters. Specifically all events need to have the following interface:

```python
process_events(events: numpy.ndarray[metavision_sdk_base._EventExtTrigger_decode | metavision_sdk_base.EventCDBuffer, buffer: numpy.ndarray[metavision_sdk_base._EventExtTrigger_decode | metavision_sdk_base.EventCDBuffer) -> None
```

This is the same interface used in most of the samples I have found on the documentation. The challenge to implement this scheme comes from the fact that there is no way in python to expand the underlying vector size, and so there is no way to fill an empty buffer of a different size.

For example:

```python
class MyFilter():
    def __init__(self) -> None:
        pass

def process_events(events, buffer):
    processed_events = # do some processing that changes the number of events
    
    # now fill the buffer with the processed events
    np.copyto(buffer.numpy(), events.numpy())

   # Error cannot broadcast shape x to y 
```

If you could resize the buffer this wouldn't be a problem. It's a simple fix, something like this to the `pod_event_buffer.h` file

```cpp
namespace Metavison {
    // existing methods ...

    // resize the underlying structure
    py::none resize(size_t size) {
        buffer_.resize(size);
        return py::none()
    } 

    std::vector<T> buffer_;

};

void export_PODEventBuffer(py::module &m, const std::string &event_buffer_name) {
    using EventBuffer = PODEventBuffer<EventType>;

    py::class_<EventBuffer, std::shared_ptr<EventBuffer>>(m, event_buffer_name.c_str(), py::buffer_protocol())
        // existing definitions ...
        .def("resize", &EventBuffer::resize, py::arg("size"),
            "resizes the buffer to the given size\n"
            "\n"
            "    :size: the size to make the buffer"
        )
}

```

lbristiel-psee, 2023-11-24:
Hi @richbai90 !
Thanks for the feedback.
That could indeed be a good addition to the SDK.
We are looking into it and will let you know when we add it.

nbourdis-psee, 2023-12-05:
Hi @richbai90, FYI we included this suggestion in the updates that will be released in the coming days, thanks for your feedback.

closed 2023-12-05 by nbourdis-psee

richbai90, 2024-01-24:
Just saw this in 4.5. Thanks!