prophesee-ai/openeb #104: Request for Real-Time Event Retrieval Support

doandv-java, 2023-12-25:
I am submitting this issue to request support for real-time event retrieval from the camera sensor in Prophesee. 

```
class EventFlow:
    event = None
    running = False
    call_number = 0
    frame_idx = 0

    def __init__(self, sensor: SensorDevice, delta_t: int = 60000):
        self.sensor = sensor
        self.mv_it = self.sensor.event_iterations(delta_t)
        self.event_list = iter(self.mv_it)
        # share data
        self.event = None
        self.running = False
        # thread
        self.read_thread = None
        self.call_number = 0

    def read_event(self):
        while self.running:
            try:
                self.call_number += 1
                self.event = next(self.event_list)
                self.frame_idx += 1
            except StopIteration as ex:
                print(ex)
                self.running = False

    def start(self):
        if self.running:
            return None
        if self.mv_it is not None:
            self.running = True
            self.read_thread = threading.Thread(target=self.read_event, name="Get Data")
            self.read_thread.daemon = True
            self.read_thread.start()

    def get_current_event(self):
        event = None
        if self.event is not None:
            event = self.event.copy()
        return event

```

I'm currently facing latency issues when attempting to obtain events from a camera sensor in real-time using Prophesee's SDK and Metavision Studio. The latency is not present when acquiring events from a raw file.

SDK Version: 4.5.0

lbristiel-psee, 2024-01-02:
Hello @doandv-java,
Latency in Studio while streaming from an event-based camera can have multiple causes. You can start by looking at [this section of the doc](https://docs.prophesee.ai/stable/metavision_studio/index.html#first-recording-from-live-camera) where we give some advices on camera installation, lighting, focus adjustement, bias settings, ROI and ESP. Another option if you want to acquire events, is to look at our various samples: [https://docs.prophesee.ai/stable/samples.html](https://docs.prophesee.ai/stable/samples.html) 
Hope this helps,
Laurent for Prophesee Support

doandv-java, 2024-01-10:
> Hello @doandv-java, Latency in Studio while streaming from an event-based camera can have multiple causes. You can start by looking at [this section of the [...]

Thanks you for comments of you