prophesee-ai/openeb #108: Incorrect Timestamp decoding in RAW files

mandulaj, 2024-02-15:
## Environment
* Ubuntu 22.04 x86_64 
* [OpenEB v4.5.2](https://github.com/prophesee-ai/openeb/releases/tag/4.5.2)
* Test file EVT3 [laser.raw](https://docs.prophesee.ai/stable/datasets.html)

## Expected Behaviour

Using the standalone `metavision_evt3_raw_file_decoder` reading the `laser.raw` file should give back the following first events:
```
x, y , p, t
656,479,0,6437104
677,479,0,6437104
679,479,0,6437104
816,479,0,6437104
885,479,0,6437104
477,479,1,6437104
386,476,0,6437104
878,476,0,6437104
665,476,1,6437104
```

## Current Behaviour

Using `metavision_core.event_io.EventsIterator` the first events are:
```
(656, 479, 0, 6384)
(677, 479, 0, 6384)
(679, 479, 0, 6384)
(816, 479, 0, 6384)
(885, 479, 0, 6384)
(477, 479, 1, 6384)
(386, 476, 0, 6384)
(878, 476, 0, 6384)
(665, 476, 1, 6384)
(350, 471, 0, 6384)
(352, 471, 0, 6384)
(355, 471, 0, 6384)
(384, 471, 0, 6384)
```

As a sidenote, using `metavision_file_info -i laser.raw` , `metavision_viewer -i laser.raw` etc results in SegFault:
```
[1]    2902066 segmentation fault (core dumped)  metavision_viewer -i laser.raw
```

## Steps to reproduce

1. Download the official [laser.raw](https://dataset.prophesee.ai/index.php/s/bX0CnMiQ2XVGGjv/download)
2. Convert it to CSV using the  `metavision_evt3_raw_file_decoder` and view first events:
```bash
metavision_evt3_raw_file_decoder laser.raw laser.csv
head laser.csv
```
3. Read the RAW using python:
```python
from metavision_core.event_io import EventsIterator
for ev in EventsIterator("laser.raw"):
    print(ev[:10])
    break
```
4. Pixel coordinates and Polarities match, but `ts` don't!

## Detailed description
The EVT3 packets are skipped until reaching the first `EVT_TIME_HIGH`. Looking at the RAW EVT3 packets:
1. `EVT_TIME_HIGH `- 0x8623 -> `0x623` =>  `(0x623 << 12) ` = 6434816
2. `EVT_TIME_LOW` -  0x68f0 -> `0x8f0` = 2288
3. TS = `(0x623 << 12) | 0x8f0` = 6437104

Which agrees with the result from `metavision_evt3_raw_file_decoder`.

The timestamp reported by `EventsIterator` is however 6384! This seems to be 4096 + `EVT_TIME_LOW` = 4096 + 2288 = 6384.
The subsequent relative timestamps are correct, so the `EVT_TIME_HIGH` is taken into account correctly, It is just subtracted from the initial timestamp and replaced by 4096.

This behavior is not documented anywhere as far as I am aware. 

I can imagine that this is done to keep the first timestamps relatively small, however, if this is indeed intended behavior, it should be documented. Otherwise, different tools will report differing timestamps.

lbristiel-psee, 2024-02-16:
Hello @mandulaj,

thanks for your very thorough report about this timestamp "situation".
Indeed, this "timestamp shifting" is done to have first timestamp relatively small.
But this is not documented at the moment and understand it can lead to some confusion.

Essentially, there are some option in the SDK to disable this "timestamp shifting" but as I was reviewing them to improve the documentation, I was having some trouble with some of them, so I need to do a complete review and make sure we have both solid code and documentation. We are working on it.

In the meantime, if you want to modify this "timestamp shifting" behaviour, the keywords to look for are:
- `do_time_shifting` in SDK Core API (initiate_device and EventsIterator, but that's were I need to double check the behaviour)
-  `do_time_shifting` in HAL API (in RawFileConfig)
- `time_shift` in SDK Driver camera class (in FileConfigHints)

Hope this helps and coming back soon with more doc,
Laurent for Prophesee Support

lbristiel-psee, 2024-03-04:
Hello @mandulaj 
We verified the behaviour of the timestamp shifting and it works OK according with out tests done on Metavision SDK 4.5.2 (and OpenEB 4.5.2). We also took this opportunity to enhance the documentation. You can now find some information about this mechanism in here: [https://docs.prophesee.ai/stable/data/streaming_decoding/timestamp_shifting.html](https://docs.prophesee.ai/stable/data/streaming_decoding/timestamp_shifting.html)
Hope this helps,
Laurent for Prophesee Support