prophesee-ai/openeb #109: CMake Error at /mnt/c/Program Files/HDF_Group/HDF5/1.14.3/cmake/hdf5-targets.cmake:42 (message)

angelsb008, 2024-02-15:
When I trying to compilated this repository to execute the comand 
"cmake .. -DBUILD_TESTING=OFF" appears the following error: 

Submodule path 'sdk/modules/driver/cpp/3rdparty/hdf5_ecf': checked out 'd3235ecf12497d57b0de80e1d989804ab2cf97a9'
CMake Error at /mnt/c/Program Files/HDF_Group/HDF5/1.14.3/cmake/hdf5-targets.cmake:42 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: hdf5::h5diff

  Targets not yet defined: hdf5::hdf5-static, hdf5::hdf5-shared,
  hdf5::mirror_server, hdf5::mirror_server_stop, hdf5::hdf5_tools-static,
  hdf5::hdf5_tools-shared, hdf5::h5diff-shared, hdf5::h5ls,
  hdf5::h5ls-shared, hdf5::h5debug, hdf5::h5repart, hdf5::h5mkgrp,
  hdf5::h5clear, hdf5::h5delete, hdf5::h5debug-shared, hdf5::h5repart-shared,
  hdf5::h5mkgrp-shared, hdf5::h5clear-shared, hdf5::h5delete-shared,
  hdf5::h5import, hdf5::h5import-shared, hdf5::h5repack,
  hdf5::h5repack-shared, hdf5::h5jam, hdf5::h5unjam, hdf5::h5jam-shared,
  hdf5::h5unjam-shared, hdf5::h5copy, hdf5::h5copy-shared, hdf5::h5stat,
  hdf5::h5stat-shared, hdf5::h5dump, hdf5::h5dump-shared,
  hdf5::h5format_convert, hdf5::h5format_convert-shared, hdf5::h5perf_serial,
  hdf5::hdf5_hl-static, hdf5::hdf5_hl-shared, hdf5::h5watch,
  hdf5::h5watch-shared, hdf5::hdf5_cpp-static, hdf5::hdf5_cpp-shared,
  hdf5::hdf5_hl_cpp-static, hdf5::hdf5_hl_cpp-shared, hdf5::hdf5_java

Call Stack (most recent call first):
  /mnt/c/Program Files/HDF_Group/HDF5/1.14.3/cmake/hdf5-config.cmake:166 (include)
  /usr/share/cmake-3.22/Modules/FindHDF5.cmake:504 (find_package)
  sdk/modules/driver/cpp/3rdparty/hdf5_ecf/CMakeLists.txt:33 (find_package)

I don't know how to fix it

lbristiel-psee, 2024-02-16:
Hello @angelsb008 

couple of questions to understand the problem and help you fix it

1) did you follow the requirements section of the installation guide?
Specially https://github.com/prophesee-ai/openeb?tab=readme-ov-file#prerequisites
Make sure that libhdf5-dev hdf5-tools were properly installed

2) can you share more information about your configs/steps so that we try to reproduce your issue
- what Operating System are you working on?
- did you clone the repo or did you download the archive?

Thanks,
Laurent for Prophesee Support

angelsb008, 2024-02-16:
Hello,

1. Yes, I follewed all the steps from the prerequisite section
2. I'm using Ubuntu 20.04 in WSL
3. I clone the repo how says in the complation section `git clone https://github.com/prophesee-ai/openeb.git --branch 4.5.2`

I would be very grateful if you could help me, thank you.

leonel-psee, 2024-02-23:
Hello @angelsb008 
Since the release 4.5.2, OpenEB is now delivered with a Dockerfile to ease the installation on Linux (see [this page](https://docs.prophesee.ai/stable/installation/linux_openeb.html#installing-dependencies) for more details)

Please could you tell if you face the same issue when installing via Docker ?

Thanks,

Leonel for Prophesee Support

lbristiel-psee, 2024-03-04:
Hi,
I hope you were finally able to fix your compilation issue either by re-trying the usual way or leveraging the Docker File  mentioned by Leonel. I am closing this issue now, come back to us if you still have some problem/questions.

closed 2024-03-04 by lbristiel-psee

boolsmith, 2024-08-22:
I ran across a similar issue. I am trying to compile on Windows 11 (I understand its not officially supported, but don't believe this is the issue here) using CMAKE. 

[image]

I have verified that hdf5 has been installed using vcpkg. 

[image]

My only other thoughts are to either find a windows 10 machine to test on, or use MS Visual Studio to build. I would prefer to try something else before I go down these rabbit holes.