prophesee-ai/openeb #111: Incorrect VECT 12-12-8 decoding

mandulaj, 2024-02-23:
# Environment
* Ubuntu 22.04 x86_64 
* [OpenEB v4.5.2](https://github.com/prophesee-ai/openeb/releases/tag/4.5.2)

* Minimal synthetic example to illustrate issue:  [example.raw.gz](https://github.com/prophesee-ai/openeb/files/14390283/example.raw.gz)

## Expected Behaviour

The example above contains the following incomplete messages:
```
0x8000 - TIME_HIGH = 0 
0x6001 - TIME_LOW = 1
0x0000 - Y = 0
0x2000 - X = 0, P = 0 -> Event(t=1, x=0,y=0, p=0)

0x300A - VECT_BASE_X X=10 P=0
0x4001 - VECT_12 - bit 0 valid -> Event(t=1, x=10,y-0,p=0)
0x4000 - VECT_12 - No valid bits

/* Missing VECT_8 */

0x80FF - TIME-HIGH = 0xFF = 255
0x2001 - X=1 P=0 -> Event(t=1044480, x=1, y=0, p=0)
```

It is not uncommon that some parts of the VECT message (such as the last VECT_8) are missing, Thus only the first valid bit events are asserted producing the following Event sequence:
```
Event(t=1, x=0,y=0, p=0)
Event(t=1, x=10,y-0,p=0)
Event(t=1044480, x=1, y=0, p=0)
```
This can be verified using the standalone `metavision_evt3_raw_file_decoder example.raw example.csv`:
```
%geometry:1280,720
0,0,0,1
10,0,0,1
1,0,0,1044480
```

## Current Behaviour
However, the decoder in the OpenEB SDK seems to not handle this correctly and instead uses the `0x80FF` packet as the last `VECT_8` marking the bits 24-31 valid. Further more, the `0x80FF`  message is then not parsed anymore and as a result the time of the successive events is incorrect. The overall result is the following CSV obtained using `metavision_file_to_csv -i example.raw`. As this is the issues with the SDK decoder, also all the other OpenEB tools and python bindings are affected. 
```
0,0,0,1     - correct
10,0,0,1   - correct
34,0,0,1   - WRONG....
35,0,0,1   
36,0,0,1
37,0,0,1
38,0,0,1
39,0,0,1
40,0,0,1
41,0,0,1     ....
1,0,0,1    - Also wrong!
```
This also seems to be the case if both the `VECT_12` and `VECT_8` message are missing.

dlzcn, 2024-02-26:
This case is not checked by the default Decoder. You will see similar behavior by setting "MV_FLAGS_EVT3_ROBUST_DECODER".  See [EVT3 Decoding Configuration](https://docs.prophesee.ai/stable/data/streaming_decoding/configuration.html?highlight=decoder#evt3-decoding-configuration)

nbourdis-psee, 2024-03-04:
hi @mandulaj, dlzcn's feedback is indeed relevant, hence I close this issue but let us know if you have further questions on this.

closed 2024-03-04 by nbourdis-psee