prophesee-ai/openeb #112: Building failed with protobuf v24.4

dlzcn, 2024-02-28:
I tried to build openeb 4.5.2 on MacOS. Following error occurred during linking,

```bash
ld: Undefined symbols:
  absl::lts_20230802::Status::UnrefNonInlined(unsigned long), referenced from:
      Metavision::save_device(Metavision::Device const&, std::__1::basic_ostream<char, std::__1::char_traits<char>>&) in camera_serialization.cpp.o
      Metavision::load_device(Metavision::Device&, std::__1::basic_istream<char, std::__1::char_traits<char>>&) in camera_serialization.cpp.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libmetavision_sdk_driver.4.5.2.dylib] Error 1
make[1]: *** [sdk/modules/driver/cpp/lib/CMakeFiles/metavision_sdk_driver.dir/all] Error 2
```

It seems libraries from `abseil` are not configured correctly.

After checking similar errors on the internet. I selected a workaround from [this issue thread](https://github.com/protocolbuffers/protobuf/issues/12637). Following is the modified `sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt`

```CMake

# Copyright (c) Prophesee S.A.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at http://www.apache.org/licenses/LICENSE-2.0
# Unless required by applicable law or agreed to in writing, software distributed under the License is distributed
# on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and limitations under the License.

if(USE_PROTOBUF)
  find_package(Protobuf REQUIRED)

    protobuf_generate_cpp(DEVICE_PROTO_SRCS DEVICE_PROTO_HDRS
        ${CMAKE_CURRENT_SOURCE_DIR}/device_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/antiflicker_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/camera_synchronization_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/digital_crop_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/digital_event_mask_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/event_rate_control_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/event_rate_noise_filter_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/event_trail_filter_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/hw_register_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/ll_biases_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/region_of_interest_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/trigger_in_state.proto
        ${CMAKE_CURRENT_SOURCE_DIR}/trigger_out_state.proto
    )

    set(PROTO_HDRS_DIRS)

    foreach(header ${DEVICE_PROTO_HDRS})
        get_filename_component(header_dir ${header} DIRECTORY)
        list(APPEND PROTO_HDRS_DIRS ${header_dir})
    endforeach(header)
endif(USE_PROTOBUF)

add_library(metavision_device_serialization_obj OBJECT ${DEVICE_PROTO_SRCS})
# The following line is needed because when linking a shared library to an object one,
# the object library needs to be compiled with -fPIC
# cf https://stackoverflow.com/questions/50600708/combining-cmake-object-libraries-with-shared-libraries
set_target_properties(metavision_device_serialization_obj
    PROPERTIES
        POSITION_INDEPENDENT_CODE ON
)

target_link_libraries(metavision_device_serialization_obj
    PRIVATE
        metavision_hal
)

if (USE_PROTOBUF)
    target_compile_definitions(metavision_device_serialization_obj PRIVATE HAS_PROTOBUF)
    target_include_directories(metavision_device_serialization_obj
        PRIVATE
            ${PROTO_HDRS_DIRS}
            ${Protobuf_INCLUDE_DIRS}
    )

    # Make it link with recent Protobuf
    string(REGEX REPLACE "^[0-9]+\.([0-9]+\.[0-9]+)$" "\\1.0" proto_libver "${Protobuf_VERSION}")
    if(proto_libver VERSION_GREATER_EQUAL "22")
        find_package(PkgConfig REQUIRED)
        pkg_check_modules(protobuf REQUIRED IMPORTED_TARGET protobuf=${proto_libver})
        target_link_libraries(metavision_device_serialization_obj PRIVATE PkgConfig::protobuf)
    else()
        # Doesn't exist for all CMake versions
        if (TARGET protobuf::libprotobuf)
            set(Protobuf_LIBRARIES protobuf::libprotobuf)
        endif ()
    
        target_link_libraries(metavision_device_serialization_obj
            PRIVATE
                ${Protobuf_LIBRARIES}
        )
    endif()
endif(USE_PROTOBUF)

```

lbristiel-psee, 2024-02-28:
Thanks for the feedback @dlzcn 
I will see if we can integrate this change in the next version of the SDK.

jthierry-psee, 2024-03-08:
@dlzcn Hi,

Could you let me know if the patch following fixes your issue?

```
diff --git a/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt b/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt
index 0bc02522c0..891e9514a1 100644
--- a/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt
+++ b/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt
@@ -8,7 +8,10 @@
 # See the License for the specific language governing permissions and limitations under the License.

 if(USE_PROTOBUF)
-    find_package(Protobuf REQUIRED)
+    find_package(Protobuf CONFIG QUIET)
+    if (NOT Protobuf_FOUND)
+        find_package(Protobuf REQUIRED)
+    endif(NOT Protobuf_FOUND)

     protobuf_generate_cpp(DEVICE_PROTO_SRCS DEVICE_PROTO_HDRS
         ${CMAKE_CURRENT_SOURCE_DIR}/device_state.proto

```

dlzcn, 2024-03-08:
@jthierry-psee I tried your patch. It doesn't work on my environment (CMake 3.27.7, Protobuf v24.4). Following is the error code during configuration.
``` CMake
-- Found HDF5: /opt/homebrew/Cellar/hdf5/1.14.2/lib/libhdf5.dylib;/opt/homebrew/opt/libaec/lib/libsz.dylib;/Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libz.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libdl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libm.tbd (found version "1.14.2")
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/usr/lib/libz.tbd (found version "1.2.12")
CMake Error at sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt:17 (protobuf_generate_cpp):
  Unknown CMake command "protobuf_generate_cpp".
```

jthierry-psee, 2024-03-11:
@dlzcn Thanks for the feedback, can you try this one please? Hopefully this would fix it.

```
diff --git a/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt b/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt
index 0bc0252..fdecdb8 100644
--- a/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt
+++ b/sdk/modules/driver/cpp/src/protobuf/CMakeLists.txt
@@ -7,10 +7,27 @@
 # on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 # See the License for the specific language governing permissions and limitations under the License.

+add_library(metavision_device_serialization_obj OBJECT)
+# The following line is needed because when linking a shared library to an object one,
+# the object library needs to be compiled with -fPIC
+# cf https://stackoverflow.com/questions/50600708/combining-cmake-object-libraries-with-shared-libraries
+set_target_properties(metavision_device_serialization_obj
+    PROPERTIES
+        POSITION_INDEPENDENT_CODE ON
+)
+
+target_link_libraries(metavision_device_serialization_obj
+    PRIVATE
+        metavision_hal
+)
+
 if(USE_PROTOBUF)
-    find_package(Protobuf REQUIRED)
+    find_package(Protobuf CONFIG QUIET)
+    if (NOT Protobuf_FOUND)
+        find_package(Protobuf REQUIRED)
+    endif(NOT Protobuf_FOUND)

-    protobuf_generate_cpp(DEVICE_PROTO_SRCS DEVICE_PROTO_HDRS
+    set(PROTO_SOURCES
         ${CMAKE_CURRENT_SOURCE_DIR}/device_state.proto
         ${CMAKE_CURRENT_SOURCE_DIR}/antiflicker_state.proto
         ${CMAKE_CURRENT_SOURCE_DIR}/camera_synchronization_state.proto
@@ -26,29 +43,21 @@ if(USE_PROTOBUF)
         ${CMAKE_CURRENT_SOURCE_DIR}/trigger_out_state.proto
     )

-    set(PROTO_HDRS_DIRS)
+    if (COMMAND protobuf_generate)
+      target_sources(metavision_device_serialization_obj PRIVATE ${PROTO_SOURCES})
+      protobuf_generate(TARGET metavision_device_serialization_obj
+                        LANGUAGE cpp
+                        OUT_VAR DEVICE_PROTO_HDRS)
+    else()
+      protobuf_generate_cpp(DEVICE_PROTO_SRCS DEVICE_PROTO_HDRS ${PROTO_SOURCES})
+      target_sources(metavision_device_serialization_obj PRIVATE ${DEVICE_PROTO_SRCS})
+    endif()

     foreach(header ${DEVICE_PROTO_HDRS})
         get_filename_component(header_dir ${header} DIRECTORY)
         list(APPEND PROTO_HDRS_DIRS ${header_dir})
     endforeach(header)
-endif(USE_PROTOBUF)
-
-add_library(metavision_device_serialization_obj OBJECT ${DEVICE_PROTO_SRCS})
-# The following line is needed because when linking a shared library to an object one,
-# the object library needs to be compiled with -fPIC
-# cf https://stackoverflow.com/questions/50600708/combining-cmake-object-libraries-with-shared-libraries
-set_target_properties(metavision_device_serialization_obj
-    PROPERTIES
-        POSITION_INDEPENDENT_CODE ON
-)
-
-target_link_libraries(metavision_device_serialization_obj
-    PRIVATE
-        metavision_hal
-)

-if (USE_PROTOBUF)
     target_compile_definitions(metavision_device_serialization_obj PRIVATE HAS_PROTOBUF)
     target_include_directories(metavision_device_serialization_obj
         PRIVATE
diff --git a/sdk/modules/driver/cpp/tests/CMakeLists.txt b/sdk/modules/driver/cpp/tests/CMakeLists.txt
index d0f66e0..ab680a6 100644
--- a/sdk/modules/driver/cpp/tests/CMakeLists.txt
+++ b/sdk/modules/driver/cpp/tests/CMakeLists.txt
@@ -60,10 +60,16 @@ add_dependencies(gtest_metavision_sdk_driver hal_dummy_test_plugin)
 # current binary dir).
 target_compile_definitions(gtest_metavision_sdk_driver PRIVATE HAL_DUMMY_TEST_PLUGIN="${CMAKE_CURRENT_BINARY_DIR}/../../../../../hal/cpp/test/plugins")

-find_package(Protobuf)
-if (Protobuf_FOUND)
-    target_compile_definitions(gtest_metavision_sdk_driver PRIVATE HAS_PROTOBUF)
-endif()
+if(USE_PROTOBUF)
+    find_package(Protobuf CONFIG QUIET)
+    if (NOT Protobuf_FOUND)
+        find_package(Protobuf)
+    endif(NOT Protobuf_FOUND)
+
+    if (Protobuf_FOUND)
+      target_compile_definitions(gtest_metavision_sdk_driver PRIVATE HAS_PROTOBUF)
+    endif()
+endif(USE_PROTOBUF)

 target_include_directories(gtest_metavision_sdk_driver PRIVATE
     ${PROJECT_SOURCE_DIR}/sdk/modules/driver/cpp/src/include

```

dlzcn, 2024-03-13:
Hi @jthierry-psee, I can confirm that this patch fixes the issue I reported! Thank you.

commit 6934dd52 (2024-03-14): [BUGFIX] Support Protobuf CMake Config Mode (fixes #112)
closed 2024-03-14 by mergify[bot]