prophesee-ai/openeb #113: Nvidia Jetson support

gyillikci, 2024-03-06:
Hello,

Is there a Nvidia Jetson support?

Thanks,

Giray

leonel-psee, 2024-03-06:
Hello Giray, 

By Nvidia Jetson support, are you refering to the ability to stream on a Nvidia Jetson platform with a Prophesee camera connected via USB ?

If yes, kindly note that the plugin source code of our EVKs are open sourced in OpenEB since [release 3.0](https://docs.prophesee.ai/stable/release_notes.html#v3-0-0-change-logs-31-05-2022). Hence, it's possible to interface our EVKs with any other platform including Nvidia Jetson and then stream, decode, display, etc... events (visit ["Open modules"](https://docs.prophesee.ai/stable/modules.html#chapter-modules-and-packaging-open) page for more details).

All the details about compilation are given [here](https://github.com/prophesee-ai/openeb).

However, some older EVK version are not anymore supported in the latest OpenEB release. For example, EVK1 since [release 4.0](https://docs.prophesee.ai/stable/release_notes.html#v4-0-0-change-logs-23-03-2023).
Also, if you have a Prophesee product with Gen4.0 sensor, you may need additional support. Please contact support@prophesee.ai in that case.

Hope this will help.

Leonel for Prophesee Support.

gyillikci, 2024-03-06:
Hello Leonel, my EVK is Prophesee Gen4.1 HD

As your reply solves this issue I am closing this and open a new issue for further question.

closed 2024-03-06 by gyillikci