prophesee-ai/openeb #115: Compile error on jetson xavier nx

mehmeteminckr, 2024-03-10:
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libavcodec.so.58: undefined reference to `v4l2_close'
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libavcodec.so.58: undefined reference to `v4l2_ioctl'
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libavcodec.so.58: undefined reference to `v4l2_open'
collect2: error: ld returned 1 exit status
make[2]: *** [sdk/modules/core/cpp/tests/CMakeFiles/gtest_metavision_sdk_core.dir/build.make:563: bin/gtest_metavision_sdk_core] Error 1
make[1]: *** [CMakeFiles/Makefile2:5066: sdk/modules/core/cpp/tests/CMakeFiles/gtest_metavision_sdk_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

I get this error while compiling on nvdia jetson xavier nx

cmake --build . --config Release -- -j `nproc`

lbristiel-psee, 2024-03-11:
Hello,

we are not familiar with this error which might be related to the platform (Jetson). You might find some information about it on NVidia related forums.
Though I see that the error happens when doing the link steps on some tests. Maybe a workaround could be to not compile the tets. This can be done with the command "cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF" as explained in https://docs.prophesee.ai/stable/installation/linux_openeb.html#compilation

Hope this helps,
Laurent for Prophesee Support

mehmeteminckr, 2024-03-12:
Hello @lbristiel-psee , I do not build tests actually as you told me in the third step. But in the 4.step(compile) i run this command 'cmake --build . --config Release -- -j `nproc`' and get this error

```
[ 85%] Built target metavision_hal_gen41_evk3_system_obj
Scanning dependencies of target metavision_hal_imx646_evk3_system_obj
Scanning dependencies of target metavision_hal_imx636_rdk2_system_obj
Scanning dependencies of target metavision_hal_genx320_cx3_system_obj
[ 86%] Building CXX object hal_psee_plugins/lib/CMakeFiles/metavision_hal_imx646_evk3_system_obj.dir/__/src/devices/imx646/imx646_tz_device.cpp.o
[ 86%] Building CXX object hal_psee_plugins/lib/CMakeFiles/metavision_hal_genx320_cx3_system_obj.dir/__/src/devices/genx320/genx320_cx3_tz_device.cpp.o
[ 86%] Building CXX object hal_psee_plugins/lib/CMakeFiles/metavision_hal_imx636_rdk2_system_obj.dir/__/src/devices/imx636/imx636_rdk2_tz_device.cpp.o
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libavcodec.so.58: undefined reference to `v4l2_close'
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libavcodec.so.58: undefined reference to `v4l2_ioctl'
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/libavcodec.so.58: undefined reference to `v4l2_open'
collect2: error: ld returned 1 exit status
make[2]: *** [sdk/modules/core/cpp/samples/metavision_csv_viewer/CMakeFiles/metavision_csv_viewer.dir/build.make:100: bin/metavision_csv_viewer] Error 1
make[1]: *** [CMakeFiles/Makefile2:3470: sdk/modules/core/cpp/samples/metavision_csv_viewer/CMakeFiles/metavision_csv_viewer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 86%] Built target metavision_hal_imx636_rdk2_system_obj
[ 86%] Built target metavision_hal_imx646_evk3_system_obj
[ 86%] Built target metavision_hal_genx320_cx3_system_obj
make: *** [Makefile:152: all] Error 2

```

With docker it seems fine, i think on jetson it is better to progress with it.

lbristiel-psee, 2024-03-13:
Thanks for the feedback.
So you are using the Dockerfile.OpenEB file mentioned in our [OpenEB install doc](https://docs.prophesee.ai/stable/installation/linux_openeb.html#installing-dependencies)?
If so, good to know that it allows you to compile on Jetson.

closed 2024-06-11 by lbristiel-psee