prophesee-ai/openeb #119: Compile error on jetson xavier

xiaozibuqi, 2024-05-25:
I want to compile OpenEB on Nvidia Jstson Xavier .
When I get to the fourth step of the code cmake ---build . --config Release -- -j `nproc`, I get an error：
Scanning dependencies of target metavision_software_info
[ 35%] Building CXX object hal_psee_plugins/lib/CMakeFiles/metavision_hal_psee_plugin_obj.dir/__/src/devices/imx646/imx646_tz_device.cpp.o
[ 36%] Building CXX object sdk/modules/base/cpp/samples/metavision_software_info/CMakeFiles/metavision_software_info.dir/metavision_software_info.cpp.o
[ 36%] Building CXX object hal/cpp/lib/CMakeFiles/metavision_hal.dir/__/src/utils/resources_folder.cpp.o
/home/nvidia/openeb/hal/cpp/src/utils/resources_folder.cpp:20:10: fatal error: filesystem: No such file or directory
 #include <filesystem>
          ^~~~~~~~~~~~
compilation terminated.
hal/cpp/lib/CMakeFiles/metavision_hal.dir/build.make:638: recipe for target 'hal/cpp/lib/CMakeFiles/metavision_hal.dir/__/src/utils/resources_folder.cpp.o' failed
make[2]: *** [hal/cpp/lib/CMakeFiles/metavision_hal.dir/__/src/utils/resources_folder.cpp.o] Error 1
Please help me what I can do to solve this problem

lbristiel-psee, 2024-06-04:
Hello,
Sorry for the delay to come back to you.
Did you find any solution?
On my side, I tested on Ubuntu amd64 (a.k.a. x64) and it compiled OK. I don't have an Nvidia Jetson Xavier at hand to reproduce the issue. Can you maybe try again with a previous version of OpenEB to see if you get the same issue?
Best,
Laurent for Prophesee Support

lbristiel-psee, 2024-06-05:
Hello again,
you can also check this other issue in which compilation on Jetson is discussed: https://github.com/prophesee-ai/openeb/issues/115
It mentions the compilation using Docker (see https://docs.prophesee.ai/stable/installation/linux_openeb.html#installing-dependencies), maybe you could give this a try as well.
Hope this helps,
Laurent

xiaozibuqi, 2024-06-05:
thanks for your help ，I'll try later

closed 2024-09-23 by lbristiel-psee