prophesee-ai/openeb #121: Failed to build OpenEB with latest Visual Studio Build Tool 2022 (17.10.1)

dlzcn, 2024-06-11:
Hello,

I recently updated Visual Studio Build Tools 2022 to the latest version (17.10.1). After the upgrade, I encountered a CMake error. Below is a screenshot of the error message. This issue can be reproduced with both openeb 4.5.2 and openeb 4.6.0 on Windows. The version of CMake that comes with this Visual Studio Build Tool is *cmake version 3.28.3-msvc11*
[image]

The file `cmake/custom_functions/add_library_version_header.cmake` needs to be updated to handle commands with spaces in paths.

lbristiel-psee, 2024-06-11:
Hello @dlzcn,

indeed, in OpenEB 4.5 and 4.6 we don't support Studio Build Tools 2022 version 17.10.
The supported/recommended version is 17.8 (Fall 2023 LTSC) though 17.9 should also work fine.

Note that we will add this information in our install guides asap.
And that we will work on supporting more recent versions of Studio in future releases.

Best,
Laurent for Prophesee Support.

ogeorget-psee, 2024-06-11:
Hi Haifeng, hope you are doing fine :) 

You're right, this is a known issue that has been patched, but yet needs to be released.

Please use this [patch](https://github.com/prophesee-ai/openeb/commit/5f7f47a0f1996247a7606b41846fde1a31b8f67a
) (or branch [issue/121](https://github.com/prophesee-ai/openeb/tree/issue/121)) while we prepare the next patch release.

closed 2024-06-11 by ogeorget-psee

dlzcn, 2024-06-12:
Hi Oliver, I am fine. Thank you for the patch. It works on my end. 

However, I had to roll back the upgrade because the new Visual Studio Build Tools (17.10.1) prevent me from mixing customized OpenEB modules with the Prophesee Metavision SDK (v4.6.0) on Windows.

lbristiel-psee, 2024-06-17:
For information, version 4.6.1 is now released and includes the fix for the issue mentioned in this ticket. Thanks again @dlzcn for your feedback!