prophesee-ai/openeb #122: Missing includes again

E1k3, 2024-06-11:
algorithm needs to be included in `hal_psee_plugins/include/boards/v4l2/v4l2_device.h`:
```
hal_psee_plugins/include/boards/v4l2/v4l2_device.h:50:27: error: ‘partition_point’ is not a member of ‘std’
auto it_pp = std::partition_point(beg, end, is_not_null);
```
Please use an IDE that detects missing includes when editing the source (it happened before https://github.com/prophesee-ai/openeb/issues/83)

lbristiel-psee, 2024-06-13:
Hi @E1k3 
Thanks for the feedback and sorry for overlooking this point.
We are currently working on mechanism to avoid that kind of issues in future versions.
For this specific missing include, it will be fixed in a 4.6.1 that should be released soon.
Best,
Laurent for Prophesee Support

closed 2024-06-13 by lbristiel-psee

lbristiel-psee, 2024-06-17:
@E1k3 for information the version 4.6.1 of OpenEB is now released and contains the fix for the issue you raised. Thanks again for the heads-up.