prophesee-ai/openeb #126: `protobuf::json::PrintOptions.always_print_primitive_field` was deprecated and was removed from the current release

E1k3, 2024-07-06:
Openeb does not compile with current protobuf. `always_print_primitive_field` probably has to be replaced with `always_print_without_presence_fields` or just removed depending on if you use proto2 or proto3 types.
See https://github.com/protocolbuffers/protobuf/pull/15674 for deprecation and https://github.com/protocolbuffers/protobuf/issues/2871 for background.

jthierry-psee, 2024-07-12:
Hi E1k3,

Yes, looks like you are correct, starting with protobuf v26.0 the field appears to have been renamed.
I opened an issue on protobuf side so they would update their online documentation.

Waiting for the PR https://github.com/prophesee-ai/openeb/pull/123 to get merged, perhaps this patch will unblock you and other users facing this issue:

```
diff --git a/sdk/modules/driver/cpp/src/camera_serialization.cpp b/sdk/modules/driver/cpp/src/camera_serialization.cpp
index b62e356..ffd214c 100644
--- a/sdk/modules/driver/cpp/src/camera_serialization.cpp
+++ b/sdk/modules/driver/cpp/src/camera_serialization.cpp
@@ -655,7 +655,11 @@ std::ostream &save_device(const Device &d, std::ostream &os) {

     google::protobuf::util::JsonPrintOptions options;
     options.add_whitespace                = true;
+#if GOOGLE_PROTOBUF_VERSION >= 5026000
+    options.always_print_without_presence_fields = true;
+#else
     options.always_print_primitive_fields = true;
+#endif
     options.preserve_proto_field_names    = true;

     std::string json;
```

E1k3, 2024-10-09:
You just released a new major version without fixing this one-liner. Please do not close issues until they are fixed.

jthierry-psee, 2024-10-11:
@E1k3 sorry about the previous wrong closing of the issue.
The fix is now merged on `main` branch.

Closing now.

closed 2024-10-11 by jthierry-psee