prophesee-ai/openeb #127: `boost::filesystem::extension` has been removed in the latest version

E1k3, 2024-09-07:
`boost::filesystem::extension` was deprecated 8 months ago and has now been removed. Please use `path.extension().string()` instead
https://www.boost.org/doc/libs/develop/libs/filesystem/doc/deprecated.html

E1k3, 2024-09-07:
```
diff --git a/sdk/modules/core/cpp/samples/metavision_player/src/main.cpp b/sdk/modules/core/cpp/samples/metavision_player/src/main.cpp
index 2df869e..9ec27d8 100644
--- a/sdk/modules/core/cpp/samples/metavision_player/src/main.cpp
+++ b/sdk/modules/core/cpp/samples/metavision_player/src/main.cpp
@@ -97,15 +97,15 @@ bool parse_command_line(int argc, const char *argv[], Parameters &app_params) {
     }

     // Check extension of provided output files
-    if (boost::filesystem::extension(app_params.out_bias_file) != ".bias") {
+    if (boost::filesystem::path(app_params.out_bias_file).extension().string() != ".bias") {
         MV_LOG_ERROR() << "Wrong extension for provided output bias file: supported extension is '.bias'";
         return false;
     }
-    if (boost::filesystem::extension(app_params.out_png_file) != ".png") {
+    if (boost::filesystem::path(app_params.out_png_file).extension().string() != ".png") {
         MV_LOG_ERROR() << "Wrong extension for provided output PNG file: supported extension is '.png'";
         return false;
     }
-    if (boost::filesystem::extension(app_params.out_avi_file) != ".avi") {
+    if (boost::filesystem::path(app_params.out_avi_file).extension().string() != ".avi") {
         MV_LOG_ERROR() << "Wrong extension for provided output AVI file: supported extension is '.avi'";
         return false;
     }
diff --git a/sdk/modules/driver/cpp/src/biases.cpp b/sdk/modules/driver/cpp/src/biases.cpp
index 13b1dbb..e1c3545 100644
--- a/sdk/modules/driver/cpp/src/biases.cpp
+++ b/sdk/modules/driver/cpp/src/biases.cpp
@@ -31,7 +31,7 @@ Biases::~Biases() {}

 void Biases::set_from_file(const std::string &biases_filename) {
     // Check extension
-    const auto extension = boost::filesystem::extension(biases_filename);
+    const auto extension = boost::filesystem::path(biases_filename).extension().string();
     if (extension != ".bias") {
         throw CameraException(CameraErrorCode::WrongExtension,
                               "For bias file '" + biases_filename +
@@ -106,7 +106,7 @@ void Biases::set_from_file(const std::string &biases_filename) {
 }

 void Biases::save_to_file(const std::string &dest_file) const {
-    const auto extension = boost::filesystem::extension(dest_file);
+    const auto extension = boost::filesystem::path(dest_file).extension().string();
     if (extension != ".bias") {
         throw CameraException(CameraErrorCode::WrongExtension,
                               "For bias file '" + dest_file +
diff --git a/sdk/modules/driver/cpp/src/camera.cpp b/sdk/modules/driver/cpp/src/camera.cpp
index 871be91..1d77514 100644
--- a/sdk/modules/driver/cpp/src/camera.cpp
+++ b/sdk/modules/driver/cpp/src/camera.cpp
@@ -298,7 +298,7 @@ bool Camera::Private::process_impl() {
 }

 bool Camera::Private::start_recording_impl(const std::string &file_path) {
-    std::string ext = boost::filesystem::extension(file_path);
+    std::string ext = boost::filesystem::path(file_path).extension().string();
     std::shared_ptr<Metavision::EventFileWriter> writer;
     if (ext == ".raw") {
         writer = std::make_shared<Metavision::RAWEventFileWriter>(file_path);
@@ -570,7 +570,7 @@ Camera Camera::from_serial(const std::string &serial, const DeviceConfig &config
 }

 Camera Camera::from_file(const std::string &file_path, const FileConfigHints &hints) {
-    if (boost::filesystem::extension(file_path) != "") {
+    if (boost::filesystem::path(file_path).extension().string() != "") {
         if (!boost::filesystem::exists(file_path)) {
             throw CameraException(CameraErrorCode::FileDoesNotExist,
                                   "Opening file at " + file_path + ": not an existing file.");
@@ -581,13 +581,13 @@ Camera Camera::from_file(const std::string &file_path, const FileConfigHints &hi
         }
[... 32 lines cut ...]
@@ -43,7 +43,7 @@ OfflineGenericPrivate::OfflineGenericPrivate(const std::string &file_path, const
     Private(detail::Config()) {
     // clang-format off
     try {
-        if (boost::filesystem::extension(file_path) == ".hdf5") {
+        if (boost::filesystem::path(file_path).extension().string() == ".hdf5") {
             file_reader_ = std::make_unique<HDF5EventFileReader>(file_path, hints.time_shift());
         } else {
             file_reader_ = std::make_unique<DATEventFileReader>(file_path);
diff --git a/sdk/modules/driver/cpp/src/camera_serialization.cpp b/sdk/modules/driver/cpp/src/camera_serialization.cpp
index b62e356..c5ecef8 100644
--- a/sdk/modules/driver/cpp/src/camera_serialization.cpp
+++ b/sdk/modules/driver/cpp/src/camera_serialization.cpp
@@ -655,7 +655,7 @@ std::ostream &save_device(const Device &d, std::ostream &os) {

     google::protobuf::util::JsonPrintOptions options;
     options.add_whitespace                = true;
-    options.always_print_primitive_fields = true;
+    options.always_print_fields_with_no_presence = true;
     options.preserve_proto_field_names    = true;

     std::string json;
```
fixes this and https://github.com/prophesee-ai/openeb/issues/126
would be better to replace the `std::string` arguments with `filesystem::path`s, but thats your decision to make.

lbristiel-psee, 2024-09-23:
Thanks for the feedback @E1k3 
We will address this in the upcoming release, which is scheduled to be available in October.

closed 2024-09-23 by lbristiel-psee

Rennylex, 2024-10-09:
Do you have a specific date for the release? thanks

nbourdis-psee, 2024-10-09:
Hi, these fixes were published with the 5.0.0 version released early this week: https://github.com/prophesee-ai/openeb/releases/tag/5.0.0