prophesee-ai/openeb #131: Unsupported OS Build

stephetw, 2024-11-19:
Hello,

I was able to successfully build openeb-5.0.0 on RHEL8. Admittedly, I am green with this, there are a few things that I modified that I do not understand what implications may be as a result. Of course, this is on an unsupported OS, so I understand if this is discarded. 
## The following are errors during the compilation step: 
```
<OPENEB_SRC_DIR>/sdk/modules/core/cpp/src/utils/video_writer.cpp:354:33: error: no matching function for call to ‘Metavision::VideoWriter::write(const cv::_InputArray&)’
     cv::VideoWriter::write(image);
```
#### Solution:
Modify <OPENEB_SRC_DIR>/sdk/modules/core/cpp/src/utils/video_writer.cpp, starting line 351:
```
#if CV_MAJOR_VERSION >= 3 && CV_MINOR_VERSION == 2
    cv::VideoWriter::write(image.getMat(-1));
#else
    cv::VideoWriter::write(image);
#endif
```
To just:
`cv::VideoWriter::write(image.getMat(-1));`

---
```
<OPENEB_SRC_DIR>/sdk/modules/core/cpp/src/utils/video_writer.cpp:377:72: error: no matching function for call to ‘Metavision::VideoWriter::operator<<(const cv::UMat&)’
     return static_cast<VideoWriter &>(cv::VideoWriter::operator<<(image));
```
#### Solution:
Modify <OPENEB_SRC_DIR>/sdk/modules/core/cpp/src/utils/video_writer.cpp, starting line 373:
```
#if CV_MAJOR_VERSION >= 3 && CV_MINOR_VERSION == 2
    MV_SDK_LOG_ERROR() << "VideoWriter :" << Metavision::Log::function << "not implemented";
    return *this;
#else
    return static_cast<VideoWriter &>(cv::VideoWriter::operator<<(image));
#endif
```
To:
```
#if CV_MAJOR_VERSION >= 3 && CV_MINOR_VERSION == 2
    MV_SDK_LOG_ERROR() << "VideoWriter :" << Metavision::Log::function << "not implemented";
    return *this;
#else
    return *this;
#endif
```
---

#### Various errors with: 
```undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'```

#### Solution:
Prepend `link_libraries(stdc++fs)` to <OPENEB_SRC_DIR>/CMakeLists.txt

---
`error: ‘setw’ is not a member of ‘std’`

In Files:
- <OPENEB_SRC_DIR>/hal/cpp/src/facilities/i_ll_biases.cpp 
- <OPENEB_SRC_DIR>/hal_psee_plugins/src/devices/genx320/genx320_roi_driver.cpp

`error: ‘put_time’ is not a member of ‘std’`

In File:
- <OPENEB_SRC_DIR>/sdk/modules/stream/cpp/src/synced_camera_system_factory.cpp

#### Solution:
Add header: `#include <iomanip>`

---

#### Package versions:
OS: Red Hat Enterprise Linux 8.10 
opencv: 3.4.6
ffmpeg: 4.2.10
HDF5: 1.10.5
libusb: 0.1.5-12
boost: 1.66.0
python: 3.11.10-1
pybind: 2.10.3-3 (precompiled in RHEL repo)
protobuf: 3.5.0-15
glew: 2.0.0-6
glfw: 3.3-2
cmake: 3.26.5-2

#### If anyone is wondering, here are the packages I installed:
`groupinstall 'Development Tools'`

`git cmake libusb libusb-devel hdf5 hdf5-devel hdf5-static boost boost-devel opencv opencv-devel ffmpeg ffmpeg-devel python3.11 python3.11-devel python3.11-pybind11-devel protobuf protobuf-devel glew-devel glfw glfw-devel`

stephetw, 2024-11-20:
Awesome thank you for the quick response!

I can confirm that this PR fixes all of the issues I saw.

---
I did try again with GCC 12.2.1, and removing link_libraries(stdc++fs). This built successfully, but there were various errors:

```
[ 23%] Building CXX object hal/cpp/lib/CMakeFiles/metavision_hal_discovery.dir/__/src/device/device_discovery.cpp.o
lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
```
```
[ 37%] Building CXX object hal/cpp/samples/metavision_platform_info/CMakeFiles/metavision_platform_info.dir/metavision_platform_info.cpp.o
lto-wrapper: warning: using serial compilation of 15 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
```
```
[ 49%] Building CXX object sdk/modules/stream/cpp/src/protobuf/CMakeFiles/metavision_device_serialization_obj.dir/trigger_out_state.pb.cc.o
In file included from /usr/include/google/protobuf/wire_format_lite_inl.h:44,
                 from /usr/include/google/protobuf/map_type_handler.h:35,
                 from /usr/include/google/protobuf/map.h:48,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /home/stephetw/prophesee-dev/openebpr/openeb/build/sdk/modules/stream/cpp/src/protobuf/region_of_interest_state.pb.h:25,
                 from /home/stephetw/prophesee-dev/openebpr/openeb/build/sdk/modules/stream/cpp/src/protobuf/region_of_interest_state.pb.cc:4:
In member function ‘void google::protobuf::internal::ElementCopier<Element, true>::operator()(Element*, const Element*, int) [with Element = bool]’,
    inlined from ‘void google::protobuf::RepeatedField<Element>::CopyArray(Element*, const Element*, int) [with Element = bool]’ at /usr/include/google/protobuf/repeated_field.h:1428:37,
    inlined from ‘void google::protobuf::RepeatedField<Element>::MoveArray(Element*, Element*, int) [with Element = bool]’ at /usr/include/google/protobuf/repeated_field.h:1422:12,
    inlined from ‘void google::protobuf::RepeatedField<Element>::Reserve(int) [with Element = bool]’ at /usr/include/google/protobuf/repeated_field.h:1403:14:
/usr/include/google/protobuf/repeated_field.h:1442:11: warning: ‘void* memcpy(void*, const void*, size_t)’ reading between 1 and 2147483647 bytes from a region of size 0 [-Wstringop-overread]
 1442 |     memcpy(to, from, static_cast<size_t>(array_size) * sizeof(Element));
      |     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
[ 51%] Building CXX object hal_psee_plugins/lib/CMakeFiles/metavision_psee_hw_layer_obj.dir/__/src/devices/common/evk2_tz_trigger_event.cpp.o
In file included from /usr/include/google/protobuf/wire_format_lite_inl.h:44,
                 from /usr/include/google/protobuf/map_type_handler.h:35,
                 from /usr/include/google/protobuf/map.h:48,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /home/stephetw/prophesee-dev/openebpr/openeb/build/sdk/modules/stream/cpp/src/protobuf/device_state.pb.h:25,
                 from /home/stephetw/prophesee-dev/openebpr/openeb/sdk/modules/stream/cpp/src/camera_serialization.cpp:34:
In member function ‘void google::protobuf::internal::ElementCopier<Element, true>::operator()(Element*, const Element*, int) [with Element = bool]’,
    inlined from ‘void google::protobuf::RepeatedField<T>::CopyArray(Element*, const Element*, int) [with Element = bool]’ at /usr/include/google/protobuf/repeated_field.h:1428:37,
    inlined from ‘void google::protobuf::RepeatedField<T>::MoveArray(Element*, Element*, int) [with Element = bool]’ at /usr/include/google/protobuf/repeated_field.h:1422:12,
    inlined from ‘void google::protobuf::RepeatedField<T>::Reserve(int) [with Element = bool]’ at /usr/include/google/protobuf/repeated_field.h:1403:14:
/usr/include/google/protobuf/repeated_field.h:1442:11: warning: ‘void* memcpy(void*, const void*, size_t)’ reading between 1 and 2147483647 bytes from a region of size 0 [-Wstringop-overread]
 1442 |     memcpy(to, from, static_cast<size_t>(array_size) * sizeof(Element));
      |     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 51%] Built target metavision_device_serialization_obj
```
```
[ 62%] Building CXX object hal/python/bindings/CMakeFiles/metavision_hal_python3_3.11.dir/i_monitoring_python.cpp.o
lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
```
```
[ 78%] Building CXX object sdk/modules/core/cpp/samples/metavision_event_frame_generation/CMakeFiles/metavision_event_frame_generation.dir/metavision_event_frame_generation.cpp.o
lto-wrapper: warning: using serial compilation of 15 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
```
```
[ 95%] Building CXX object hal_psee_plugins/lib/CMakeFiles/metavision_hal_psee_plugin_obj.dir/__/src/devices/imx636/imx636_evk2_tz_device.cpp.o
lto-wrapper: warning: using serial compilation of 8 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
```

jthierry-psee, 2024-11-21:
@stephetw Perfect, thanks for confirming. The PR was merged. Closing this issue.

closed 2024-11-21 by jthierry-psee