prophesee-ai/openeb #132: Failed to install Python modules in host OS ( = NOT virtualized Python env).

s-tory, 2024-11-20:
hi, thank you to dev good hard and sdk!

## My env
- CPU arch: x64 (Intel Core i7)
- Host OS: Ubuntu 24.04.1 LTS

## Occured issue
- Build and install by following commands **WITHOUT virtualized Python env** (ex. venv, anaconda, miniforge).
```
sudo apt update
sudo apt -y install apt-utils build-essential software-properties-common wget unzip curl git cmake
sudo apt -y install libopencv-dev libboost-all-dev libusb-1.0-0-dev libprotobuf-dev protobuf-compiler
sudo apt -y install libhdf5-dev hdf5-tools libglew-dev libglfw3-dev libcanberra-gtk-module ffmpeg

cd
wget https://github.com/pybind/pybind11/archive/v2.11.0.zip
unzip v2.11.0.zip
cd pybind11-2.11.0/
mkdir build && cd build
cmake .. -DPYBIND11_TEST=OFF
cmake --build .
sudo cmake --build . --target install

cd
git clone -b 5.0.0 https://github.com/prophesee-ai/openeb.git
cd openeb
mkdir build
cd build
cmake .. -DBUILD_TESTING=OFF
cmake --build . --config Release -- -j 4
sudo cmake --build . --target install
```
- Has installed OpenEB SDK (shared objects and) Python modules in `/usr/local/local/lib/python3.12/dist-packages/` 
  - **`/local` is doubled! That path is incorrect! So following Python interaction has been occured.**
```
$ python3
Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import metavision_sdk_core
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'metavision_sdk_core'
```

## Reference
- Build and install OpenEB SDK **WITH virtualized Python env = miniforge.**
```
wget "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
bash Miniforge3-$(uname)-$(uname -m).sh
```
- Has installed OpenEB SDK Python modules in `/home/{user_name}/miniforge3/lib/python3.12/site-packages/` 
  - **No problem in following Python interaction.**
```
(base) $ python
Python 3.12.7 | packaged by conda-forge | (main, Oct  4 2024, 16:05:46) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import metavision_sdk_core
>>> 
```

jthierry-psee, 2024-11-26:
@s-tory , thank you for reporting the issue. I was able to reproduce the issue.
Can you confirm whether this PR fixes the issue on your end?
https://github.com/prophesee-ai/openeb/pull/134

Thanks,

s-tory, 2024-11-27:
@jthierry-psee
Perfect! This issue has been fixed exactly. Thank you very much.
```
cd
git clone -b 5.0.0 https://github.com/prophesee-ai/openeb.git
cd openeb
mkdir build
cd build
cmake .. -DBUILD_TESTING=OFF
cmake --build . --config Release -- -j 4
sudo cmake --build . --target install
```
>> install_manifest.txt
```
...
/usr/local/lib/python3.12/dist-packages/metavision_hal.cpython-312-x86_64-linux-gnu.so
...
/usr/local/lib/python3.12/dist-packages/metavision_sdk_base.cpython-312-x86_64-linux-gnu.so
...
/usr/local/lib/python3.12/dist-packages/metavision_sdk_core.cpython-312-x86_64-linux-gnu.so
...
/usr/local/lib/python3.12/dist-packages/metavision_core/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/events_iterator.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/event_bufferizer.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/adaptive_rate_events_iterator.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/py_reader.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/meta_event_producer.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/raw_info.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/npy_tools.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/box_npy_reader.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/event_frame_iterator.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/dat_tools.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/h5_io.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/raw_reader.py
/usr/local/lib/python3.12/dist-packages/metavision_core/event_io/live_replay.py
...
/usr/local/lib/python3.12/dist-packages/metavision_sdk_stream.cpython-312-x86_64-linux-gnu.so
...
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/corner_video_stream_dataset.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/data_module.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/gpu_corner_esim.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/utils.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/lightning_model.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/compute_homography_reprojection_error.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/corner_tracker.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/corner_detection/firenet.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/gpu_simulator.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/cutoff_kernels.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/single_image_make_events_cpu.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/events_kernels.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/simulator.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/video_stream_dataset.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/video_stream_dataset_with_events_cpu.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/video_to_event/simu_events_iterator.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/files.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/show_or_write.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/train_utils.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/event_utils.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/color_utils.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/utils/torch_ops.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/image_planar_motion_stream.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/video_stream.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/stream_dataloader.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/scheduling.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/raw_image_planar_motion_stream.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/corner_planar_motion_stream.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/data/camera_poses.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/event_to_video/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/event_to_video/data_module.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/event_to_video/event_to_video.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/event_to_video/lightning_model.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/event_to_video/gpu_esim.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/core/__init__.py
[... cut ...]
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/losses/warp.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/losses/perceptual_loss.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/preprocessing/__init__.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/preprocessing/event_to_tensor_torch.py
/usr/local/lib/python3.12/dist-packages/metavision_core_ml/preprocessing/viz.py
...
/usr/local/lib/python3.12/dist-packages/metavision_sdk_ui.cpython-312-x86_64-linux-gnu.so
```

jthierry-psee, 2024-11-27:
@s-tory Thanks for confirming. I merged the PR. I'll close this issue now.

closed 2024-11-27 by jthierry-psee