prophesee-ai/openeb #138: Does not build against opencv 4.11

E1k3, 2025-01-20:
```
openeb/sdk/modules/core/cpp/include/metavision/sdk/core/algorithms/roi_mask_algorithm.h:129:34: error: no matching function for call to ‘cv::Rect_<int>::contains(<brace-enclosed initializer list>) const’
```

Fix by replacing
`return rectangle.contains({x, y});`
with
`return rectangle.contains(cv::Point{x, y});`

lbristiel-psee, 2025-01-21:
Thank you for the feedback! You’re right ,OpenEB is not currently compatible with OpenCV 4.11 as-is, and the change you mentioned would indeed be necessary to ensure compatibility. However, the installation guide for Ubuntu 22 and 24 specifies using the default OpenCV installation, which installs an older version (prior to 4.11). On those platforms, OpenEB compiles and functions as expected without issues. But your message will be useful for those trying to work with version 4.11.

E1k3, 2025-01-21:
Not everyone is using Ubuntu and my suggested change does not break compatibility with older OpenCV versions, so why not just change one line and make it work for everyone?

lbristiel-psee, 2025-02-04:
Hi,
The change was applied in this PR: https://github.com/prophesee-ai/openeb/pull/139 
Thanks again for the feedback.
Laurent

closed 2025-02-04 by lbristiel-psee