prophesee-ai/openeb #141: Cannot find metavision_ml library for installation

kmanninen, 2025-01-31:
Hello, I inherited a python application that appears to reference a legacy metavision library (metavision_ml). Specifically, it uses the "histo" function as documented [here](https://docs.prophesee.ai/stable/api/python/ml/preprocessing.html#metavision_ml.preprocessing.event_to_tensor.histo).

I am running my application on:

Ubuntu v24.04.1 LTS (noble)
Python version 3.11.11

I have followed the installation instructions as documented [here](https://docs.prophesee.ai/stable/installation/linux_openeb_with_packages.html#installation) and I cannot seem to find any reference to the "metavision_ml" library in my system. I can see metavision_core_ml and metavision_core subdirectories in my /usr/lib/python3/dist-packages folder. I can also import functions from metavision_core_ml, metavision_sdk_core, and metavision_core. But my system has no knowledge of metavision_ml.

Did I miss an installation step?

Thank you.

kmanninen, 2025-01-31:
I forgot to add the version of metavision-sdk-core that I installed. Here is the output from the dpkg command:

```
dpkg -s metavision-sdk-core-ml-python
Package: metavision-sdk-core-ml-python
Status: install ok installed
Priority: optional
Section: devel
Installed-Size: 288
Maintainer: support@prophesee.ai
Architecture: amd64
Version: 5.1.0
Depends: metavision-sdk-core-python3.12 (= 5.1.0), metavision-sdk-base-python3.12 (= 5.1.0)
Description: Metavision SDK CORE ML Python Modules.
 License: Copyright (c) Prophesee S.A. - All Rights Reserved
Homepage: https://support.prophesee.ai
```

lbristiel-psee, 2025-01-31:
Hello @kmanninen 

The metavision_ml library is part of the "advanced module" ML of the SDK. And OpenEB contains only the "Open modules" of the SDK. There is an explanation of the module organization with regard to the packagint in that doc page: https://docs.prophesee.ai/stable/modules.html

There was also a pricing/packaging change since version SDK 5.0 so metavision_ml module is only distributed to paid customer (the only free option is OpenEB which does not contain metavision_ml). So if your code is compatible with the previous version of the SDK, you should go for this one as it was free and contained metavision_ml. This version is SDK 4.6 and can be downloaded by filling this form: https://www.prophesee.ai/metavision-intelligence-sdk-download/

I hope this clarifies and helps,

The metavision_ml library is part of the Advanced ML Module in the SDK, whereas OpenEB includes only the Open Modules of the SDK. You can find a detailed explanation of the module organization and packaging structure on this documentation page: [https://docs.prophesee.ai/stable/modules.html](https://docs.prophesee.ai/stable/modules.html)

Additionally, since SDK version 5.0, there has been a pricing and packaging change, meaning that the metavision_ml module is now only available to paid customers. The only free option is OpenEB, which does not include metavision_ml.

If your code is compatible with an earlier SDK version, you may want to use SDK 4.6, as it was free and included metavision_ml. You can download it by filling out this form: [https://www.prophesee.ai/metavision-intelligence-sdk-download/](https://www.prophesee.ai/metavision-intelligence-sdk-download/)

I hope this clarifies things

Best,
Laurent for Prophesee Support

kmanninen, 2025-01-31:
Thank you, Laurent, for the clear explanation! Yes, this helps.

It appears that if I want to use SDK 4.6 I need to downgrade my OS from Ubuntu 24 to Ubuntu 20 or Ubuntu 22, correct?

Thank you again.

lbristiel-psee, 2025-01-31:
Yes indeed, SDK 4.6.2 is available only for Ubuntu 20 and 22.

closed 2025-02-07 by lbristiel-psee