prophesee-ai/openeb #143: BUILD_TESTING OFF Does not disable all tests.

Tetragramm, 2025-02-11:
Specifically, the python tests, such as 
[metavision-hal-python-bindings-test](https://github.com/prophesee-ai/openeb/blob/8be27a37c6f2de9d1c38d2d7177c6680b35b577a/hal/python/tests/CMakeLists.txt#L12) and [pytests_core](https://github.com/prophesee-ai/openeb/blob/8be27a37c6f2de9d1c38d2d7177c6680b35b577a/sdk/modules/core/python/tests/CMakeLists.txt) as two examples.

Should be a simple matter of adding the same if-condition to these tests, but there are at least 6, possibly more.

lbristiel-psee, 2025-02-19:
Hello @Tetragramm 

In the OpenEB repository, the BUILD_TESTING option controls whether the test suite is compiled or not, but the tests themselves are not executed during compilation, regardless of whether BUILD_TESTING is set to ON or OFF.

To actually run the tests, you need to manually execute the following command "ctest -C Release" as documented here: [https://docs.prophesee.ai/stable/installation/linux_openeb.html#running-tests](https://docs.prophesee.ai/stable/installation/linux_openeb.html#running-tests)

Could you provide more details about the issue you're encountering?
- What command are you running?
- What behavior do you expect?
- What actually happens? (How do you observe that unexpected tests are being launched?)

This will help us better understand and diagnose the problem.

Thanks,
Laurent for Prophesee Support

Tetragramm, 2025-02-19:
OpenEB is integrated as a CMake subproject in my project.
BUILD_TESTING is set to OFF
COMPILE_PYTHON3_BINDINGS is set to ON

A combination of VSCode and CMake gathers up all the tests created (both mine and those from OpenEB) and places them into the list of unit tests, where they fail.

[image]

Here is one of the places tests are being incorrectly generated:
https://github.com/prophesee-ai/openeb/blob/b3160f0421b9ef08917a27cc2c24cc46705ddcb4/sdk/modules/core_ml/python/CMakeLists.txt#L20
Here is a place it is being done correctly:
https://github.com/prophesee-ai/openeb/blob/b3160f0421b9ef08917a27cc2c24cc46705ddcb4/sdk/modules/base/cpp/CMakeLists.txt#L26

lbristiel-psee, 2025-02-24:
Got it, thanks for the detailed feedback!
We will incorporate the BUILD_TESTING changes in the next release of SDK/OpenEB.