prophesee-ai/openeb #145: example of Metavision::Histo3dDecoder use?

skal-hs, 2025-02-18:
Hi,

i'm trying to use the Histo3dDecoder with:

```c++
 std::unique_ptr<Metavision::Device> device = Metavision::DeviceDiscovery::open("");
 auto* decoder = device->get_facility<Metavision::Histo3dDecoder>();
``` 

but i get this compile error:

```c++
/usr/local/include/metavision/hal/device/device.h:32:23: error: static assertion failed due to requirement 'std::is_base_of<Metavision::I_RegistrableFacility<Metavision::Histo3dDecoder, void>, Metavision::Histo3dDecoder>::value': Unable to get facility of unregistrable facility type.
   32 |         static_assert(std::is_base_of<I_RegistrableFacility<FacilityType>, FacilityType>::value,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
camera_tuner.cpp:191:41: note: in instantiation of function template specialization 'Metavision::Device::get_facility<Metavision::Histo3dDecoder>' requested here
  191 |     auto* i_histogram_decoder = device->get_facility<Metavision::Histo3dDecoder>();
```

Is it expected? What code am i missing here?
Is there any example of EHCDecoder usage somewhere?

Thanks a lot!
skal/

lbristiel-psee, 2025-02-20:
Hello @skal-hs 

the proper way to instanciate the Histo3D events decoder is the following:
```
#include <metavision/sdk/base/events/raw_event_frame_histo.h>
#include <metavision/hal/facilities/i_event_frame_decoder.h>
auto* decoder = device->get_facility<Metavision::I_EventFrameDecoder<Metavision::RawEventFrameHisto>>();
```

The only example we have in the SDK that leverage this decoder at the moment is within the Stream module:
[https://github.com/prophesee-ai/openeb/blob/main/sdk/modules/stream/cpp/src/camera_live.cpp#L247](https://github.com/prophesee-ai/openeb/blob/main/sdk/modules/stream/cpp/src/camera_live.cpp#L247)

Hope this helps,
Laurent for Prophesee Support

skal-hs, 2025-02-20:
Hi Laurent,

thanks for the pointers. 

I've tried to use it similarly to the test [`psee_raw_file_decoder_gtest.cpp`](https://github.com/prophesee-ai/openeb/blob/main/hal_psee_plugins/test/psee_raw_file_decoder_gtest.cpp#L1152)
This test passes ok, and the same code transposed to my app works too.

But if i replace  the test's code `device = DeviceDiscovery::open_raw_file("./histo3d.raw", cfg);` by the one actually opening the real camera: `device = DeviceDiscovery::open(serial);`, then it stops working. 
The callback dealing with the histogram is not called.

-> Is the `I_EventFrameDecoder<RawEventFrameHisto>` facility really using the hardware EHC, or is just suitable for histo3d RAW files? 
And if the latter, how can i acces the real hardware EHC histograms?

thanks!
skal/

lbristiel-psee, 2025-02-21:
Hi,

EHC is only available with the GenX320 sensor ([https://www.prophesee.ai/event-based-sensor-genx320/](https://www.prophesee.ai/event-based-sensor-genx320/)). If you own a camera that does not contain this sensor (but rather an [IMX636](https://www.prophesee.ai/event-based-sensor-imx636-sony-prophesee/) for example), then no ECH is present and no Histo3D data will be produced.

If you do own a GenX320 based product, then you can ask access to Prophesee Knowledge Center ([https://www.prophesee.ai/resources-access-request/](https://www.prophesee.ai/resources-access-request/)) where will can access the GenX320 Datasheet and User Manual that explain how to configure EHC.

Best,
Laurent for Prophesee Support