prophesee-ai/openeb #147: Python packages couldn't be imported( for lack of .dll?)

GourmandChau, 2025-03-29:
Hi, sorry to bother you. I followed your tutorial to employ OpenEB in my PC and run the test suite successfully. Metaviasion_viewer works. However, when I go further to get started using Python API, the packages just couldn't be imported as "ImportError: DLL load failed". To be specific, the package "metavision_sdk_base_paths_internal" can be imported normally while others fail.
Through dumpin, I checked the dependencies of all .pyd files, and I'm pretty sure all .dll dependencies can be found in <OPENEB_SRC_DIR>/build/bin/Release. And I also set the environment viarables.

Windows 10
and Here is the information of the build files writing:

-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.20348.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.38.33145.0
-- The CXX compiler identification is MSVC 19.38.33145.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/VisualStudio/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/VisualStudio/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for a CUDA compiler
-- Looking for a CUDA compiler - NOTFOUND
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- Found LibUSB: D:/vcpkg/installed/x64-windows/include/libusb-1.0
-- Found Boost: D:/vcpkg/installed/x64-windows/include (found version "1.78.0") found components: program_options timer chrono thread atomic
-- Found Python3: D:/Python_venv/prophesee/Scripts/python.exe (found version "3.12.8") found components: Interpreter Development Development.Module Development.Embed
-- Performing Test HAS_MSVC_GL_LTCG
-- Performing Test HAS_MSVC_GL_LTCG - Success
-- Found pybind11: D:/vcpkg/installed/x64-windows/include (found version "2.12.0")
-- Found Protobuf: D:/vcpkg/installed/x64-windows/tools/protobuf/protoc.exe (found version "3.21.12.0")
-- Found TIFF: optimized;D:/vcpkg/installed/x64-windows/lib/tiff.lib;debug;D:/vcpkg/installed/x64-windows/debug/lib/tiffd.lib (found version "4.6.0")
-- Found OpenCV: D:/vcpkg/installed/x64-windows (found version "4.8.0") found components: core highgui imgproc videoio imgcodecs calib3d objdetect
-- Found HDF5: optimized;D:/vcpkg/installed/x64-windows/lib/hdf5_cpp.lib;debug;D:/vcpkg/installed/x64-windows/debug/lib/hdf5_cpp_D.lib;optimized;D:/vcpkg/installed/x64-windows/lib/hdf5.lib;debug;D:/vcpkg/installed/x64-windows/debug/lib/hdf5_D.lib (found suitable version "1.14.2", minimum required is "1.10.2") found components: CXX
-- Building METAVISION_SDK_MODULES_OPEN modules : base;core;stream;core_ml;ui
-- Building METAVISION_SDK_MODULES_ADVANCED modules : analytics;cv;ml;calibration;cv3d
-- METAVISION_SELECTED_MODULES=base;core;stream;core_ml;ui;analytics;cv;ml;calibration;cv3d
-- METAVISION_SDK_MODULES_AVAILABLE=base;core;stream;core_ml;ui
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- Found V4l2 header : v4l2_headers-NOTFOUND
-- Found dma-buf header : dma_buf_header-NOTFOUND
-- Found dma-heap header : dma_heap_header-NOTFOUND
CMake Deprecation Warning at hal_psee_plugins/lib/CMakeLists.txt:191 (cmake_policy):
  The OLD behavior for policy CMP0095 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Generating internal bias files.
--  - vdac18_8m1.calib
--  - gen3_idac.calib
--  - gen3_idac_n_thick.calib
--  - gen3_idac_n_thin.calib
--  - gen3_idac_p_thick.calib
--  - gen3_idac_p_thin.calib
--  - gen3_idac_railp.calib
-- Found HDF5: hdf5::hdf5-shared (found version "1.14.2")
-- Found OpenGL: opengl32
-- Found GLEW: D:/vcpkg/installed/x64-windows/share/glew/glew-config.cmake
-- Configuring done (8.0s)
-- Generating done (0.7s)
-- Build files have been written to: D:/openeb/openeb/build

I'm not sure whether I miss some installation steps, but .exe works successfully. Could you plz have a look? If more information is needed, let me know.

Thank you so much!

lbristiel-psee, 2025-03-31:
Hi,

Can you confirm you did not have any compilation error?
(the output you show is the output of "cmake .." not the "cmake --build .")

Though this must be a pythonpath or similar issue where your python executable is not able to find the SDK packages. So maybe the deploy step was not properly executed. In the deploy step of 
 the compilation ([https://docs.prophesee.ai/stable/installation/windows_openeb.html#compilation-using-cmake](https://docs.prophesee.ai/stable/installation/windows_openeb.html#compilation-using-cmake)) which option did you follow: 1 or 2 ? Did you do all the steps of your option?

To help you find the root cause you can also look at this FAQ entry:
[https://docs.prophesee.ai/stable/faq.html#how-can-i-resolve-a-modulenotfounderror-when-running-a-python-script-or-importing-a-python-sdk-module](https://docs.prophesee.ai/stable/faq.html#how-can-i-resolve-a-modulenotfounderror-when-running-a-python-script-or-importing-a-python-sdk-module)

Laurent for Prophesee Support

GourmandChau, 2025-03-31:
Hi, firstly I'd like to appreciate for your kind answer, and here is my further information.

> Can you confirm you did not have any compilation error?
> (the output you show is the output of "cmake .." not the "cmake --build .")

I checked the compilation output row by row and there are no any dirrect errors but a few warnings --- I'm not sure if they do make any weird influence --- and for convinience, I paste the output of the command "cmake --build" in output.txt if you'd like to have a check(It's too long so I didn't paste it directly in my comment).

[output.txt](https://github.com/user-attachments/files/19536409/output.txt)

> Though this must be a pythonpath or similar issue where your python executable is not able to find the SDK packages. So maybe the deploy step was not properly executed. In the deploy step of
> the [...]

I followed option 1, and yes, I did all the steps of the option.
But for the packges that couldn't be imported, my description might be ambiguous. To be clear, the problem is that the python executable is able to find the SDK packages, but some Classes like Camera, CameraStreamSlicer just couldn't be found. 

For instance, I open the __init__.py in metavision_sdk_stream_internal, and as you'll see in example.png, the package **_metavision_sdk_base_paths_internal_** is totally able to be imported while others aren't. 

[image]
[image]

By the way, I tried to re-employ OpenEB but this problem still exists.

Thanks for your patience!

lbristiel-psee, 2025-04-04:
Hi @GourmandChau 

thanks for the feedback. The warning in the compilation process can be ignored. Now for the packages that can not be imported, you are sharing screenshots of your IDE but the error could be linked to IDE configuration somehow. Could you share what error you get when you actually try to launch the Python sample from a command line? And also from the same commane line, please show the value of PYTHONPATH.

Laurent

GourmandChau, 2025-04-04:
Thanks for your continuous attention! Now I turn to use your Metavision SDK 4.6 and It seems to work properly when I import the packages. However, The error still shows the same when I make the following command: python D:\events_camera\main.py -i D:\openeb\datasets\laser.raw
Here is the details:

[HAL][WARNING]     no plugin found 
[HAL][ERROR] While opening RAW file 'D:\openeb\datasets\laser.raw':

Traceback (most recent call last):
  File "D:\events_camera\main.py", line 29, in <module>
    main()
  File "D:\events_camera\main.py", line 22, in main
    mv_iterator = EventsIterator(input_path=args.event_file_path, delta_t=1000)
  File "D:\Prophesee\lib\python3\site-packages\metavision_core\event_io\events_iterator.py", line 70, in __init__
    self._init_readers(input_path=input_path, **kwargs)
  File "D:\Prophesee\lib\python3\site-packages\metavision_core\event_io\events_iterator.py", line 90, in _init_readers
    self.reader = RawReaderBase(input_path, delta_t=self.delta_t, ev_count=self.n_events, **kwargs)
  File "D:\Prophesee\lib\python3\site-packages\metavision_core\event_io\raw_reader.py", line 95, in __init__
    self.reset()
  File "D:\Prophesee\lib\python3\site-packages\metavision_core\event_io\raw_reader.py", line 188, in reset
    self.device = initiate_device(self.path, do_time_shifting=self.do_time_shifting,
  File "D:\Prophesee\lib\python3\site-packages\metavision_core\event_io\raw_reader.py", line 53, in initiate_device
    device = DeviceDiscovery.open_raw_file(path, file_stream_config)
RuntimeError: 
------------------------------------------------
Metavision HAL exception

Error 101000: No plugin available for input stream. Could not identify source from header:
% Date 2020-09-25 07:48:30
% camera_integrator_name Prophesee
% evt 3.0
% firmware_version 3.2.3
% integrator_name Prophesee
% plugin_integrator_name Prophesee
% plugin_name hal_plugin_gen41_evk3
% serial_number 00001034
% system_ID 48
% end

------------------------------------------------

I followed the tutorial (http://docs.prophesee.ai/4.6.2/installation/windows_openeb.html) to set my env variables:

[image]

[image]

And added those  in PATH:

[image]

I've read the FAQ (https://docs.prophesee.ai/4.6.2/faq.html#how-can-i-fix-the-error-no-plugin-available-for-input-stream-could-not-identify-source-from-header-when-reading-raw-files). And there are 2 listed solutions. 
The first one is to set the MV_HAL_PLUGIN_PATH and as you can see, I did set this variable.
The second one is to check its header. However, this file is downloaded from your samples(https://docs.prophesee.ai/4.6.2/datasets.html#chapter-datasets).

Therefore, I wondered if there was something wrong with the hal package first.

But now I'm confused again = =. Could you help me?

Thanks a lot!

GourmandChau, 2025-04-04:
I forgot to attach the files in MV_HAL_PLUGIN_PATH, and they are listed as followed:

[image]

Is there anything missed in this path?

GourmandChau, 2025-04-05:
Thanks a lot for your support! I've just solved this problem.

It might be that the vcpkg path has set in a wrong way. Originally, I set D:\vcpkg in my PATH, where the whole vcpkg installed. Then I found the .dll dependencies of metavision_sdk_core_internal.cp39-win_amd64.pyd through dumpbin and realized that the dependencies in vcpkg, such as opencv_core4.dll, might not be found. 

Therefore, I set D:\vcpkg\installed\x64-windows\bin in my PATH and successfully run the code.

closed 2025-04-05 by GourmandChau