prophesee-ai/openeb #149: Memory Growth Issue

294978174, 2025-04-08:
I'm using OpenEB 5.1.0. When running the demo from sdk/core/cpp_samples/metavision_composed_viewer/metavision_composed_viewer.cpp, I observed continuous memory growth.
This might be caused by these lines:

const auto left_image_ref = frame_composer.add_new_subimage_parameters(0, 0, {width, height}, Metavision::FrameComposer::GrayToColorOptions());
Metavision::PeriodicFrameGenerationAlgorithm left_frame_generator(width, height, accumulation_time_ms * 1000, 66.0);
cam.cd().add_callback([&left_frame_generator](const Metavision::EventCD *begin, const Metavision::EventCD *end)
                      { left_frame_generator.process_events(begin, end); });
left_frame_generator.set_output_callback(
    [&left_image_ref, &frame_composer](Metavision::timestamp t, cv::Mat &frame_data)
    {
        if (!frame_data.empty())
        {
            frame_composer.update_subimage(left_image_ref, frame_data);
        }
    });

But strangely, when I tested on another ​​identical machine​​, the memory issue didn't occur.
​​Test results across four machines:​​

3 machines showed persistent memory growth
1 machine worked normally

The system environment is approximately: ​​Ubuntu 22.04, CUDA 11.7, FFmpeg 5.01​​.

lbristiel-psee, 2025-04-08:
Hello

The memory growth you're seeing is likely due to a high event rate from the camera, causing PeriodicFrameGenerationAlgorithm to accumulate more data than expected, especially if the processing can't keep up.

The fact that one machine doesn't show the issue suggests the difference comes from sensor conditions, not the code.

To reduce memory usage, we recommend:
- Tune lighting: avoid flickering or overly bright sources.
- Adjust focus: a blurry image can generate excess noise.
- Set sensor biases: lower sensitivity using BiasTuner or via code.
- Simplify the scene: test with a static or low-motion target.

You can find some information about those points in this section of the doc: https://docs.prophesee.ai/stable/metavision_studio/index.html#first-recording-from-a-live-camera (it is the doc of Studio which is not part of OpenEB but it applies anyway).

Hope this helps,
Laurent for Prophesee Support

closed 2025-04-08 by lbristiel-psee

294978174, 2025-04-09:
My scenario involves unavoidable strong indoor lighting. Even with proper focus and adjustments to the Event Trail Filter, overexposure persists, resulting in excessive noise. Do you have any recommendations for mitigating this issue?

lbristiel-psee, 2025-04-10:
@294978174 you should tune the biases to be less sentitive to light.
See 
- [https://www.youtube.com/watch?v=7c7n5DMuatY](https://www.youtube.com/watch?v=7c7n5DMuatY)
- [https://docs.prophesee.ai/stable/hw/manuals/biases.html](https://docs.prophesee.ai/stable/hw/manuals/biases.html)

294978174, 2025-04-10:
Great, thanks!